espressif/llvm-project #3: Undefined reference when using atomics

lexxvir, 2020-01-15:
Hi,

I encountered into issue when try to use Rust compiler's `AtomicBool`.

Linker error: `undefined reference to __sync_lock_test_and_set_1`.
Quite same error for `AtomicUsize`: `undefined reference to __sync_lock_test_and_set_4`.

It seems that LLVM somehow should replace these symbols by real atomic instructions for xtensa target.

vgarleanu, 2020-04-29:
pinging @lexxvir and @pacmancoder . Whats the status on this? I am still getting this linking error when using rustc

lexxvir, 2020-04-29:
Hi @vgarleanu !

I just tested it again, on recent compiler (see #20) and it's OK. Code links successfully and runs without issue. I my test I just incrementing AtomicUsize in the loop by the two threads and output the values.

vgarleanu, 2020-04-29:
@lexxvir Thanks a lot for taking your time to test it out on your system. I just did some more testing myself and it seems that only `AtomicBool` causes a linker error throwing `undefined reference to `__sync_lock_test_and_set_1'`. `AtomicUsize` works fine. Do you mind testing AtomicBool on your side? FYI my snippet is just this `rs
#[no_mangle]
pub fn app_main() {
    static FIRST_PANIC: AtomicBool = AtomicBool::new(true);
    FIRST_PANIC.swap(false, SeqCst);
}
`

lexxvir, 2020-04-29:
@vgarleanu Yeah, you're right, `swap` leads to the linker error for me too:
`__sync_lock_test_and_set_1` for `AtomicBool` and  `__sync_lock_test_and_set_4` for `AtomicUsize`.

Just in case, my test:
[atomic.zip](https://github.com/espressif/llvm-project/files/4551312/atomic.zip)

vgarleanu, 2020-04-29:
@lexxvir Alright, how do we go about tracking the cause down? Is there any maintainer from espressif that we can ping in?

lexxvir, 2020-04-29:
@vgarleanu I think @andreisfr or @igrr could help here...

vgarleanu, 2020-04-29:
Should i perhaps open a separate issue? Im unsure if this bug relates with the one atop.

reitermarkus, 2020-05-02:
@vgarleanu, I think we are all talking about the same bug. I can confirm this bug still exists in the newest version. I am still using the C shim @pacmancoder mentioned as a workaround for now.

andreisfr, 2020-05-06:
@lexxvir , @vgarleanu , thank you for information, we investigate the problem.

vgarleanu, 2020-06-06:
hey @andreisfr whats the status on this?

lexxvir, 2020-12-13:
Hi there,

I've just built new Rust compiler using [xtensa_release_11.0.0](https://github.com/MabezDev/rust-xtensa/pull/43) branch and it seems compiler now has atomics!
To check that I've used test from https://github.com/espressif/llvm-project/issues/3#issuecomment-621135637.

vgarleanu, 2020-12-22:
Gotcha, ill check everything on my side sometime this week!

gerekon, 2021-05-18:
Closed as it was fixed by https://github.com/espressif/llvm-project/tree/xtensa_release_11.0.0

closed 2021-05-18 by gerekon