espressif/llvm-project #4: Tracking Issue for merging to upstream

polarathene, 2020-01-15:
I understand it will take quite some time for this backend to be merged upstream into LLVM. 

This issue is for notifying anyone that subscribes to it when the merge to upstream has completed closing this issue.

igrr, 2020-01-15:
We are moving this and other issues to https://github.com/espressif/llvm-project, which will host the Xtensa patches for LLVM based on the new monorepo layout. This new repository will be updated with Xtensa patches based on LLVM 9 soon.

andreisfr, 2020-02-10:
@polarathene, @DarkWiiPlayer , @breckwagner,  we updated current patches (1-10), which are in LLVM review, according to changes in upstream version and reviewers requests for changes.

lexxvir, 2020-07-22:
Hi @andreisfr,

Is there any plan/way to merge this fork into the upstream?

I've skimmed some pull requests on the reviews.llvm.org, it seems the process is stalled.

andreisfr, 2020-07-22:
Hi @lexxvir,

Yes, we plan to continue integration process of the Xtensa backend to the upstream.

windoze, 2020-09-08:
Is there any progress for this? 
I'm dying to see it merging into upstream, it's really a PITA to build the whole LLVM/Clang project from the scratch in a CI pipeline.

shymega, 2021-05-19:
Hey, hope you don't mind me chasing this up :smile: I've been getting ready to start work on Rust code for an ESP32, but without official LLVM and Rust support, I fear I'll have to use C. It does take ages to compile as well. I might be able to influence a different chip, but I think the ESP32 is a solid choice.

Thanks.

torpak, 2021-06-25:
It seems nobody at LLVM is even answering since may. Does anyone know a reason for this?

clouds56, 2021-06-26:
@torpak according to this https://reviews.llvm.org/D64830#2615149 comment, I think they are waiting the author to rewrite a RFC since things changed a lot.
Correct me if wrong.

Hao123br, 2021-06-26:
@clouds56 the author already sent another RFC. https://lists.llvm.org/pipermail/llvm-dev/2021-March/149090.html

clouds56, 2021-06-26:
@Hao123br thanks for the link, shall we also provide the link in the thread and ping aykevl & tstellar, maybe they are too busy and missing the RFC?

saeidakbari, 2022-07-09:
Is any new news on reviewing process?

flaminisx, 2022-09-21:
just noticed that patches 4, 5, 6, 8 and 10 are also accepted

igrr, 2022-09-26:
For future reference, the new RFC is at https://discourse.llvm.org/t/rfc-request-for-upstream-tensilica-xtensa-esp32-backend/65355.

SergioGasquez, 2022-12-13:
> > The patches are in review (see links in https://esp32.com/viewtopic.php?t=11412#p46446) and further timeline will depend on the review process. We will keep updating the patch set to address the feedback!
> 
> From Espressif Forum post [Re: Status of LLVM support?](https://esp32.com/viewtopic.php?t=11412#p46446)
> 
> > * [x]  [D64826: [Xtensa 1/10] Recognize Xtensa in triple parsing code.](https://reviews.llvm.org/D64826)
> > * [x]  [D64827: [Xtensa 2/10] Add Xtensa ELF definitions.](https://reviews.llvm.org/D64827)
> > * [x]  [D64829: [Xtensa 3/10] Add initial version of the Xtensa backend.](https://reviews.llvm.org/D64829)
> > * [x]  [D64830: [Xtensa 4/10] Add basic *td files with Xtensa architecture description.](https://reviews.llvm.org/D64830)
> > * [x]  [D64831: [Xtensa 5/10] Add Xtensa MCTargetDescr initial functionality.](https://reviews.llvm.org/D64831)
> > * [x]  [D64832: [Xtensa 6/10] Add Xtensa basic assembler parser.](https://reviews.llvm.org/D64832)
> > * [ ]  [D64833: [Xtensa 7/10] Add Xtensa instruction printer.](https://reviews.llvm.org/D64833)
> > * [x]  [D64834: [Xtensa 8/10] Add support of the Xtensa shift/load/store/move and processor control instructions.](https://reviews.llvm.org/D64834)
> > * [ ]  [D64835: [Xtensa 9/10] Add basic support of Xtensa disassembler.](https://reviews.llvm.org/D64835)
> > * [x]  [D64836: [Xtensa 10/10] Add relaxations and fixups. Add rest part of Xtensa Core Instructions.](https://reviews.llvm.org/D64836)
> 
> Status as of 2019-12-24 [x] = Accepted [ ] = Needs Review

Patches 7 and 9 have now been accepted!

shish, 2022-12-31:
Looks like these are all merged into the main branch now, if I'm reading it right? Merry Christmas all :D

aq1018, 2023-03-17:
Should this be closed? LLVM has has accepted all patches according to this article - https://www.phoronix.com/news/LLVM-Xtensa-Backend

Skgland, 2023-03-18:
Unless https://github.com/esp-rs/rust/issues/89#issuecomment-1426982176 is outdated, there are still some parts missing.

igrr, 2023-03-27:
The next set of patches has been uploaded to LLVM Phabricator. I am adding the list of patches in the same format as before. We'll be updating this comment as patches are accepted.

* [ ] [D145658: [Xtensa] Initial support of the ALU operations.](https://reviews.llvm.org/D145658)
* [ ] [D145660: [Xtensa] Codegen support for memory operations](https://reviews.llvm.org/D145660)
* [ ] [D145661: [Xtensa] Add Constant Pool](https://reviews.llvm.org/D145661)
* [ ] [D145662: [Xtensa] Implement asm representation of the Constant Pool.](https://reviews.llvm.org/D145662)
* [ ] [D145663: [Xtensa] Implement lowering constants.](https://reviews.llvm.org/D145663)
* [ ] [D145664: [Xtensa] Add support of the Xtensa function calls](https://reviews.llvm.org/D145664)
* [ ] [D145665: [Xtensa] Implement lowering ConstantPool and address operations.](https://reviews.llvm.org/D145665)
* [ ] [D145666: [Xtensa] Implement emitPrologue/emitEpilogue](https://reviews.llvm.org/D145666)
* [ ] [D145667: [Xtensa] Lower stack operations.](https://reviews.llvm.org/D145667)
* [ ] [D145668: [Xtensa] Implement lowering SELECT\_CC, SETCC](https://reviews.llvm.org/D145668)
* [ ] [D146948: [Xtensa] Support for a variety of additional LLVM IR constructs.](https://reviews.llvm.org/D146948)
* [ ] [D146949: [Xtensa] Lower SHIFT PARTS and shift operations.](https://reviews.llvm.org/D146949)
* [ ] [D146950: [Xtensa] Implement load pseudo operations and patterns.](https://reviews.llvm.org/D146950)
* [ ] [D146951: [Xtensa] Support for variable arguments](https://reviews.llvm.org/D146951)
* [ ] [D146952: [Xtensa] Implement lowering BR\_JT operation](https://reviews.llvm.org/D146952)
* [ ] [D146953: [Xtensa] Support for address intrinsics.](https://reviews.llvm.org/D146953)
* [ ] [D146954: [Xtensa] Add basic support for inline asm constraints](https://reviews.llvm.org/D146954)
* [ ] [D146955: [Xtensa] Implement volatile load/store.](https://reviews.llvm.org/D146955)
* [ ] [D146956: [Xtensa] Implement branch analysis](https://reviews.llvm.org/D146956)
* [ ] [D146957: [Xtensa] Implement support for the BranchRelaxation](https://reviews.llvm.org/D146957)

orowith2os, 2023-06-28:
@igrr @andreisfr reading some of the reviews, a common occurrence is the lack of tests. How's that going? It's been a few months, so figured I'd ask.

andreisfr, 2023-06-28:
@orowith2os , we preparing update for patches, including tests, soon it will be uploaded to Phabricator.

Skgland, 2023-09-02:
It appears that the llvm-project is movin/has moved from Phabricator to GitHub (<https://github.com/llvm/llvm-project/pull/65162>) and Phabricator is suppost to go read-only next month/in October. I guess the patches need to be resubmitted via GitHub at some point.

teburd, 2024-02-15:
Very cool, definitely looking forward to seeing this make its way in

alhirzel, 2024-02-15:
For those keeping track, it looks like [a recently-merged PR](https://github.com/llvm/llvm-project/pull/78548) covers the first five of the above-mentioned patchsets:

> * [x]  [D145658: [Xtensa] Initial support of the ALU operations.](https://reviews.llvm.org/D145658)
> * [x]  [D145660: [Xtensa] Codegen support for memory operations](https://reviews.llvm.org/D145660)
> * [x]  [D145661: [Xtensa] Add Constant Pool](https://reviews.llvm.org/D145661)
> * [x]  [D145662: [Xtensa] Implement asm representation of the Constant Pool.](https://reviews.llvm.org/D145662)
> * [x]  [D145663: [Xtensa] Implement lowering constants.](https://reviews.llvm.org/D145663)
>  * [ ]  [D145664: [Xtensa] Add support of the Xtensa function calls](https://reviews.llvm.org/D145664)
>  * [ ]  [D145665: [Xtensa] Implement lowering ConstantPool and address operations.](https://reviews.llvm.org/D145665)
>  * [ ]  [D145666: [Xtensa] Implement emitPrologue/emitEpilogue](https://reviews.llvm.org/D145666)
>  * [ ]  [D145667: [Xtensa] Lower stack operations.](https://reviews.llvm.org/D145667)
>  * [ ]  [D145668: [Xtensa] Implement lowering SELECT_CC, SETCC](https://reviews.llvm.org/D145668)
>  * [ ]  [D146948: [Xtensa] Support for a variety of additional LLVM IR constructs.](https://reviews.llvm.org/D146948)
>  * [ ]  [D146949: [Xtensa] Lower SHIFT PARTS and shift operations.](https://reviews.llvm.org/D146949)
>  * [ ]  [D146950: [Xtensa] Implement load pseudo operations and patterns.](https://reviews.llvm.org/D146950)
>  * [ ]  [D146951: [Xtensa] Support for variable arguments](https://reviews.llvm.org/D146951)
>  * [ ]  [D146952: [Xtensa] Implement lowering BR_JT operation](https://reviews.llvm.org/D146952)
>  * [ ]  [D146953: [Xtensa] Support for address intrinsics.](https://reviews.llvm.org/D146953)
>  * [ ]  [D146954: [Xtensa] Add basic support for inline asm constraints](https://reviews.llvm.org/D146954)
>  * [ ]  [D146955: [Xtensa] Implement volatile load/store.](https://reviews.llvm.org/D146955)
>  * [ ]  [D146956: [Xtensa] Implement branch analysis](https://reviews.llvm.org/D146956)
>  * [ ]  [D146957: [Xtensa] Implement support for the BranchRelaxation](https://reviews.llvm.org/D146957)

andreisfr, 2024-02-15:
@alhirzel ,  recent  PR contains only 1st mentioned patch. We removed 4 patches from PR, according to reviewers comments. But we preparing next patch now)

alhirzel, 2024-02-15:
> @alhirzel , recent PR contains only 1st mentioned patch. We removed 4 patches from PR, according to reviewers comments. But we preparing next patch now)

Gotcha - edited to correct this. Sorry!

gerekon, 2024-02-16:
Hi guys!

LLVM upstreaming process has moved to GitHub and looks like original patches posted on Phabricator are going to be restructured. I suppose that in the most cases they will be split into pieces.

We decided to avoid posting a lot of patches at the same time in order to ease review and update process. So my suggestion is to update the list and include only patches which has respective PR on GitHub. So the actual list should look like below. And we will update it with new PRs.

- [x] [[Xtensa] Initial codegen support from IR](https://github.com/llvm/llvm-project/pull/78548). (Part of [D145658: [Xtensa] Initial support of the ALU operations.](https://reviews.llvm.org/D145658))

@andreisfr @sstefan1 WDYT?

andreisfr, 2024-02-16:
@gerekon, I think yes, the new series of patches will be different from the previous one

gerekon, 2024-02-16:
@andreisfr Could you mention this PR in every upstream PR? In order to make tracking easier

andreisfr, 2024-02-16:
> @andreisfr Could you mention this PR in every upstream PR? In order to make tracking easier

yes

gerekon, 2024-03-01:
Upstream patches list update
- [x] [[Xtensa] Initial codegen support from IR](https://github.com/llvm/llvm-project/pull/78548). (Split from [D145658: [Xtensa] Initial support of the ALU operations.](https://reviews.llvm.org/D145658))
- [ ] [[Xtensa] Implement base CallConvention](https://github.com/llvm/llvm-project/pull/83280).

SergioGasquez, 2024-04-24:
https://github.com/llvm/llvm-project/pull/83280 is already merged! :tada:

gerekon, 2024-04-24:
:smiley: 
Upstream patches list update
- [x] [[Xtensa] Initial codegen support from IR](https://github.com/llvm/llvm-project/pull/78548). (Split from [D145658: [Xtensa] Initial support of the ALU operations.](https://reviews.llvm.org/D145658))
- [x] [[Xtensa] Implement base CallConvention](https://github.com/llvm/llvm-project/pull/83280).

gerekon, 2024-05-23:
Upstream patches list update. We pushed new patch to upstream
- [x] [[Xtensa] Initial codegen support from IR](https://github.com/llvm/llvm-project/pull/78548). (Split from [D145658: [Xtensa] Initial support of the ALU operations.](https://reviews.llvm.org/D145658))
- [x] [[Xtensa] Implement base CallConvention](https://github.com/llvm/llvm-project/pull/83280).
- [ ] [[Xtensa] Implement FrameLowering methods and stack operation lowering](https://github.com/llvm/llvm-project/pull/92960)

andreisfr, 2024-06-12:
Upstream patches list update

- [ ] https://github.com/llvm/llvm-project/pull/78548. (Split from [D145658: [Xtensa] Initial support of the ALU operations.](https://reviews.llvm.org/D145658))
- [ ]  https://github.com/llvm/llvm-project/pull/83280.
- [ ]  https://github.com/llvm/llvm-project/pull/92960
- [ ] https://github.com/llvm/llvm-project/pull/95256

tom-borcin, 2024-06-25:
@SergioGasquez There are 13 patches left:

1. Implement lowering SELECT_CC, SETCC  - In Progress
2. Lower stack operations.
3. Support for a variety of additional LLVM IR constructs.
4. Lower SHIFT PARTS and shift operations.
5. Implement load pseudo operations and patterns.
6. Support for variable arguments
7. Implement lowering BR_JT operation
8. Support for address intrinsics.
9. Add basic support for inline asm constraints
10. Implement volatile load/store.
11. Implement branch analysis
12. Implement support for the BranchRelaxation

gerekon, 2024-06-25:
@tom-borcin These are the patches initially posted to Phabricator. I think that was just some part of our patches.  After LLVM development moved to GH the structure of patches has changed so the list above is not correct.
I think it is better to ask @andreisfr about the list of patches.

andreisfr, 2024-06-26:
@SergioGasquez , @tom-borcin , 
Probably we need to merge about 30-40 patches to implement most of functionality to be able generate Rust code and compile ESP-IDF examples:
1) Currently we have 10 prepared patches with base functionality not upstreamed (https://github.com/espressif/llvm-project/issues/4#issuecomment-2189374398). This patches are not very large, so I think optimistic estimation is to spend 1-1,5 week on each patch to upstream. 
So process in optimal way will be finished in September. 
2) Implement other Xtensa features like Window, Boolean, Float, etc. It is about 8-12 patches. Also in optimistic case will take another 3 month and wiil be finished in November. 
3) Implement Xtensa targets in Clang esp32,esp8266, etc" and add support for multilib and linker. It is about 8-12 patches. Some of these patches will be complex, so in optimistic case will take another 2-3 month and will be finished in January/Febraury. 
In a pessimistic case, this process may take another 2-4 months.

JetForMe, 2024-06-27:
Pity it can't be sooner! What's the main obstacle? Getting upstream to accept the patches?

nlgranger, 2024-06-28:
I would hazard a guess that it's about upstream finding time to review the patches.

andreisfr, 2024-06-28:
- [ ] 1. https://github.com/llvm/llvm-project/pull/78548. (Split from [D145658: [Xtensa] Initial support of the ALU operations.]
- [ ] 2. https://reviews.llvm.org/D145658))
- [ ] 3.  https://github.com/llvm/llvm-project/pull/83280.
- [ ] 4.  https://github.com/llvm/llvm-project/pull/92960
- [ ] 5.  https://github.com/llvm/llvm-project/pull/95256
- [ ] 6. https://github.com/llvm/llvm-project/pull/97017

andreisfr, 2024-06-28:
@JetForMe , @nlgranger delay in pull requests merging is that we must to update commits according to reviewers comments,  this is iterative process and also currently only a few reviewers commenting our PRs. Maybe, If you have the opportunity to involve more code reviewers, that would be great.

andreisfr, 2024-06-28:
@bugadani , we have about 10-15 patches, which depends on each other. I think,  after merging these patches we will be able to create several PR at the same time.

aykevl, 2024-06-28:
@andreisfr one way to interest more people in reviewing Xtensa patches might be to review patches from other people. Not sure if you are doing this already, but it is usually very much appreciated. When they see you reviewing their code, they might be more likely to review Xtensa patches.

andreisfr, 2024-06-28:
@aykevl , thank you for advice!

JetForMe, 2024-06-29:
I will try to bring this to the attention of the Embedded Swift folks at Apple. They currently support the ESP32-C6, and I think very much want to encourage Matter adoption, so convincing them that Xtensa support is important might help speed things up.

I suspect Espressif has their own connections to Apple, at the level of the Matter standards development. It would be good to shake that tree, too.

mortenbekditlevsen, 2024-06-29:
@JetForMe Will you do this through the Swift Evolution Forum@

@andreisfr In case you're wondering about the spike in interest for these patches, the Swift programming language quite recently added an 'embedded' subset of the language. 
This is why myself and many others are now following this work with much interest. 😊

MihaelIsaev, 2024-06-29:
I'm subscribed to all the activity in this repository because waiting for Xtensa support to start using ESP32 with Embedded Swift. Would love to support a thread about that on Swift Forums 😊 there are [already at least one thread](https://forums.swift.org/t/is-there-support-for-xtensa-lx7-cores/72458) and [on reddit](https://www.reddit.com/r/esp32/comments/1ddjbbk/embedded_swift_on_esp32/) as well

teburd, 2024-06-29:
Are the hifi patches also upstreamed?

gerekon, 2024-06-29:
@teburd 
> Are the hifi patches also upstreamed?

Do you mean ones from [this PR](https://github.com/espressif/llvm-project/pull/85)?
Not yet I think. They need Boolean and Floating point features as far as I see.

bugadani, 2024-06-29:
Considering how long it takes to upstream few-hundered-LOC PRs, I am somewhat worried about throwing 80k+ LOC patches in 😅

gerekon, 2024-06-29:
> Considering how long it takes to upstream few-hundered-LOC PRs, I am somewhat worried about throwing 80k+ LOC patches in 😅

I expect that after the next 10-15 patches we will start doing this in parallel. Maybe we can try to keep 2-3 active patches in upstream review and see if it will speed up the process.

@andreisfr what do you think?

andreisfr, 2024-07-01:
@gerekon , yes, I think we need to upstream about 10-15 patches and probably we will be able to create parallel PRs

ProfFan, 2024-07-31:
Is there an update or target timeline?

johnnyasantoss, 2024-08-22:
Apparently https://github.com/llvm/llvm-project/pull/99981 just got merged

JetForMe, 2024-08-22:
IIUC everything is merged? I wonder how long before it makes it into a Swift toolchain.

Skgland, 2024-08-22:
> IIUC everything is merged?

No, https://github.com/espressif/llvm-project/issues/4#issuecomment-2260035936 only includes the seven patches already submitted to upstream. Based on https://github.com/espressif/llvm-project/issues/4#issuecomment-2201145347 that makes it 7 out of 10-15 patches that need to be submitted in series after which more patches will follow in parallel.

JetForMe, 2024-08-22:
[This comment](https://github.com/espressif/llvm-project/issues/4#issuecomment-2189374398) listed the following items remain. I was unable to easily correlate them to patches in later comments.

> @SergioGasquez There are 13 patches left:
> 
> 1. Implement lowering SELECT_CC, SETCC  - In Progress
> 2. Add support of the Xtensa function calls
> 3. Lower stack operations.
> 4. Support for a variety of additional LLVM IR constructs.
> 5. Lower SHIFT PARTS and shift operations.
> 6. Implement load pseudo operations and patterns.
> 7. Support for variable arguments
> 8. Implement lowering BR_JT operation
> 9. Support for address intrinsics.
> 10. Add basic support for inline asm constraints
> 11. Implement volatile load/store.
> 12. Implement branch analysis
> 13. Implement support for the BranchRelaxation

andreisfr, 2024-08-26:
Current status. 

- [ ]  1. https://github.com/llvm/llvm-project/pull/78548. (Split from [D145658: [Xtensa] Initial support of the ALU operations.]
- [ ]  2. https://reviews.llvm.org/D145658))
- [ ]  3. https://github.com/llvm/llvm-project/pull/83280.
- [ ]  4. https://github.com/llvm/llvm-project/pull/92960
- [ ]  5. https://github.com/llvm/llvm-project/pull/95256
- [ ]  6. https://github.com/llvm/llvm-project/pull/97017
- [ ]  7. https://github.com/llvm/llvm-project/pull/99981
- [ ]  8. https://github.com/llvm/llvm-project/pull/106053

TurkeyMan, 2024-08-28:
🎉 

What happens next now that these all seem to be done?

bugadani, 2024-08-28:
Now the next patch will be submitted, then the next, and the next... :)

[16 more replies]