espressif/llvm-project #5: Support for LLVM 8/9

sapir, 2020-01-15:
I'd like to use this with the current Rust compiler, but Rust currently depends on LLVM 8 or 9.

I got this fork compiling with LLVM 8 here:
https://github.com/sapir/llvm-project/tree/llvm-8-xtensa

I think it works, as I managed to use it to compile an example Rust project [as described here](https://quickhack.net/nom/blog/2019-05-14-build-rust-environment-for-esp32.html), although I haven't actually run the resulting binary on an ESP32 yet.

andreisfr, 2020-01-24:
Hi @sapir , @beriberikix , @jeandudey , @aykevl ! Now current version of the LLVM Xtensa backend is based on release 9.0.1 (default branch in https://github.com/espressif/llvm-project).

jklmnn, 2020-01-24:
I tried to build it and `llc` showed me `xtensa` as available target although it is not clear to me how to actually build an xtensa object file with this. Running `llvm/llvm-obj/bin/llc -march=xtensa -o test.s test.bc` seems to create the same file as it does for x86. Is there any documentation on how to use this backend to build object files for this architecture?

james-callahan, 2020-02-24:
@andreisfr sounds like this issue can be closed then?

andreisfr, 2020-03-21:
Seems issues is fixed

closed 2020-03-23 by gerekon