espressif/llvm-project #8: insertIndirectBranch not implemented but SizeRelaxationPass uses it

bjoernQ, 2020-01-15:
I'm using a Rust port (https://github.com/MabezDev/rust-xtensa) that uses this LLVM backend. It works on my OSX system but not on my Windows system - but it used to work on Windows using some quite early commits of this LLVM backend. For me actually it crashes during compilation on Windows.

I investigated a bit and was able to track it down to _`insertIndirectBranch` isn't implemented_ 

Then I realized that `insertIndirectBranch` is called via BranchRelaxation which is enabled by https://github.com/espressif/llvm-xtensa/blob/757e18f722dbdcd98b8479e25041b1eee1128ce9/lib/Target/Xtensa/XtensaTargetMachine.cpp#L137

Probably it used to work for me before since previously XtensaTargetMachine configured the passes like this

```
  addPass(createXtensaBranchSelectionPass());
  addPass(createXtensaSizeReductionPass());
```

So it wasn't using BranchRelaxation before.

I don't understand why it's working on OSX and apparently Linux and it's only me who runs into this problem (and only on Windows) - however given the code it shouldn't be platform dependent.

I guess implementing `insertIndirectBranch` would be a good solution. 

Locally I just commented out `addPass(&BranchRelaxationPassID);` from `XtensaPassConfig::addPreEmitPass()` and I'm now able to compile and run code. 
Unfortunately implementing a proper solution (i.e. `insertIndirectBranch`) myself is way over my head.

andreisfr, 2020-03-22:
Hi @bjoernQ ! Please, check https://github.com/espressif/llvm-project/tree/xtensa_release_9.0.1 version, I suppose this version fixes problem.

bjoernQ, 2020-03-23:
I will give it a try and soon as possible. Thanks for the update

bjoernQ, 2020-04-09:
Works fine for me! Thanks!

closed 2020-04-09 by bjoernQ