espressif/llvm-project #10: Incorrect version?

MabezDev, 2020-01-16:
I've built the xtensa_release_9.x branch but when I run `--version` on any binaries they report `10.0.0svn`, this breaks the build for [my rust fork](https://github.com/MabezDev/rust-xtensa) as the compiler interacts with llvm differently depending on the version see [PassWrapper.cpp](https://github.com/rust-lang/rust/blob/117ceeba400e69aa7b8ba5155acd1c8c0cdf7f7f/src/rustllvm/PassWrapper.cpp#L536). 

Full output from `llc --version`
```bash
LLVM (http://llvm.org/):
  LLVM version 10.0.0svn
  Optimized build.
  Default target: x86_64-unknown-linux-gnu
  Host CPU: znver1

  Registered Targets:
    x86    - 32-bit X86: Pentium-Pro and above
    x86-64 - 64-bit X86: EM64T and AMD64
    xtensa - Xtensa 32
```

andreisfr, 2020-01-17:
Hi. @MabezDev ! Thank you for information. The xtensa_release_9.x branch is based on llvm-project master branch with latest commit https://github.com/llvm/llvm-project/commit/0f30960619f9c6b3566d3d4ed9a9bdd6ee0d3725#diff-b7c61103e5b04a057783630466888503  .  So, on 30 September release_9.x was just published and release_10.x was only in plans. Do you have principal restrictions to use exactly release_9.x(not release_10.x) in Rust fork?

MabezDev, 2020-01-17:
> Do you have principal restrictions to use exactly release_9.x(not release_10.x) in Rust fork?

When building the compiler, it uses the output from `llvm-config --version` to determine the version of llvm. I have manually hard coded this value and the build still fails.

Rust has recently landed the changes required for the llvm 10 upgrade in https://github.com/rust-lang/rust/pull/67900 but like you said this fork is based of a commit in master after the 9 release. This means the fork doesn't have all the required upstream changes for use with the Rust compiler (and I suspect any compiler looking to use LTO).

Perhaps we should rebase this fork from the 10.x release?

Great to see the fork is coming along nicely though! Is there a list of features it currently supports? I.e Can it do codegen now or does it still rely on an external assembler?

gerekon, 2020-01-17:
> Perhaps we should rebase this fork from the 10.x release?

@MabezDev Is this critical for you to work with `release_10.x` fork? AFAIK LLVM10 is not released yet.
So we'd prefer to fork from stable LLVM9 release 9.0.1 which is 30 days old. When the first stable LLVM10 will be released we will make a fork for it too.

andreisfr, 2020-01-17:
> Great to see the fork is coming along nicely though! Is there a list of features it currently supports? I.e Can it do codegen now or does it still rely on an external assembler?

The xtensa_release_9.x can generate object files, it uses binutils assembler only for *.S input files. Also added base support of disassembler and corrected some reported issues with atomic operations, indirect branches and other code improvements.

MabezDev, 2020-01-17:
> @MabezDev Is this critical for you to work with release_10.x fork? AFAIK LLVM10 is not released yet.
So we'd prefer to fork from stable LLVM9 release 9.0.1 which is 30 days old. When the first stable LLVM10 will be released we will make a fork for it too.

No, in fact I'd much rather it was LLVM 9 actually. Forking from the stable 9.x release branch sounds like the best approach.

> The xtensa_release_9.x can generate object files

Awesome! I think this will solve a number of issues on my end, will report back once I get it working.

> Also added base support of disassembler and corrected some reported issues with atomic operations, indirect branches and other code improvements.

Excellent, I've had a few users reporting similiar issues

MabezDev, 2020-01-18:
@andreisfr FYI, locally I rebased onto the 9.x release and it works brilliantly.

I now have full debug info in Rust! I can step through, set breakpoints, inspect variables etc

[image]

Which will finally close #6 :tada: 

To clarify, is the plan to rebase this fork onto 9.0.1 stable release?

andreisfr, 2020-01-18:
@MabezDev , good news, thanks) 
> To clarify, is the plan to rebase this fork onto 9.0.1 stable release?

Yes, we now rebase this fork to 9.0.1

igrr, 2020-01-24:
The default branch of this repository is now based on release/9.x (9.0.1)

closed 2020-01-24 by igrr