espressif/llvm-project #13: LLVM ERROR: Regression from previous version of the fork

reitermarkus, 2020-01-27:
Hi,

I noticed a regression when compiling a project using my fork of `rust-xtensa` here: https://github.com/reitermarkus/rust/tree/xtensa-support

When compiling a project containing

```rust
std::net::ToSocketAddrs::to_socket_addrs("0.0.0.0:80");
```

I get the following error:

```
LLVM ERROR: Error while trying to spill A8 from class AR: Cannot scavenge register without an emergency spill slot!
```

This compiled fine previously with the `llvm-xtensa` fork.

Not sure how to debug this further.

andreisfr, 2020-01-27:
Hi @reitermarkus! Thank you for information! Can you attach the code which causes an error to reproduce situation? For example like there https://github.com/espressif/llvm-xtensa/issues/4#issuecomment-481116928

reitermarkus, 2020-01-28:
Hi, @andreisfr, here is the relevant code:

[llvm-spill-error.zip](https://github.com/espressif/llvm-project/files/4120302/llvm-spill-error.zip)

reitermarkus, 2020-01-28:
I just noticed that this error only occurs when `lto = true`, with `lto = false` or `lto = "thin"` it compiles without problems.

andreisfr, 2020-01-28:
Thank you very much @reitermarkus . Could you also generate llvm IR file (*.ll) with option "-emit-llvm", please? Probably with IR file we will be able to reproduce problem faster.

reitermarkus, 2020-01-29:
Here is the IR file:

[llvm_spill_error-e1c6646996d6d257.llvm_spill_error.23cs0faf-cgu.0.rcgu.ll.zip](https://github.com/espressif/llvm-project/files/4125917/llvm_spill_error-e1c6646996d6d257.llvm_spill_error.23cs0faf-cgu.0.rcgu.ll.zip)

closed 2020-02-14 by espressif-bot

canufeel, 2020-04-15:
Had the same error while compiling Rust project using `rust-xtensa`. It failed to compile `core` crate. Removing `codegen-units = 1` from `Cargo.toml` did the trick for me. `lto` is presently set to true.

andreisfr, 2020-04-15:
Hi, @canufeel ! Did you use latest xtensa_release_9.0.1 branch? At least it should fix "spill errors" reported by @reitermarkus in test case.

canufeel, 2020-04-16:
Hi @andreisfr I was using the tip of `xtensa-target` branch for `rust-xtensa` which points to this llvm if I understand this correctly - https://github.com/rust-lang/rust/compare/master...MabezDev:xtensa-target#diff-8903239df476d7401cf9e76af0252622R43 . Then i just setup the quickstart project as it is described here - https://github.com/MabezDev/xtensa-rust-quickstart and tried to compile it. The only thing i have changed about the project is the target which i set to `xtensa-esp8266-none-elf`. Here is stack trace from it:

```
xargo build
warning: Patch `rustc-std-workspace-alloc v1.99.0 (/path/to/project/rust-xtensa/src/tools/rustc-std-workspace-alloc)` was not used in the crate graph.
Patch `rustc-std-workspace-core v1.99.0 (/path/to/project/rust-xtensa/src/tools/rustc-std-workspace-core)` was not used in the crate graph.
Patch `rustc-std-workspace-std v1.99.0 (/path/to/project/rust-xtensa/src/tools/rustc-std-workspace-std)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling core v0.0.0 (/path/to/project/rust-xtensa/src/libcore)
LLVM ERROR: Error while trying to spill A8 from class AR: Cannot scavenge register without an emergency spill slot!
error: could not compile `core`.

To learn more, run the command again with --verbose.
error: `"cargo" "build" "--release" "--manifest-path" "/var/folders/gk/5lqbgvpd2gq52m91c3d71w900000gn/T/xargo.FggBzlih9NdZ/Cargo.toml" "--target" "xtensa-esp8266-none-elf" "-p" "core"` failed with exit code: Some(101)
   0: backtrace::backtrace::trace
   1: backtrace::capture::Backtrace::new
   2: error_chain::make_backtrace
   3: <error_chain::State as core::default::Default>::default
   4: <std::process::Command as xargo::extensions::CommandExt>::run
   5: xargo::sysroot::build
   6: xargo::sysroot::update
   7: xargo::main_inner
   8: std::rt::lang_start::{{closure}}
   9: std::panicking::try::do_call
  10: __rust_maybe_catch_panic
  11: std::rt::lang_start_internal
  12: main
```

MabezDev, 2020-10-04:
Hi @andreisfr I have randomly seen this cropping up still, but it only seems to affect the esp8266 target (at least, I haven't seen it building for the esp32).

I've just ran into this again:

```rust
LLVM ERROR: Error while trying to spill A2 from class AR: Cannot scavenge register without an emergency spill slot!
error: could not compile `core`
```

Here is the llvm-ir, which I hope is of use: [core-090545185f2e882a.core.5zurvqyc-cgu.0.rcgu.ll.zip](https://github.com/espressif/llvm-project/files/5323169/core-090545185f2e882a.core.5zurvqyc-cgu.0.rcgu.ll.zip)

commit da018456 (2020-02-14): [Xtensa] Corrected spill slot allocation.
Closes #13