espressif/llvm-project #18: Support for LLVM 10

aykevl, 2020-03-16:
Perhaps a bit premature, but LLVM 10 can be released any moment now: RC 4 has just been tagged:
https://lists.llvm.org/pipermail/llvm-dev/2020-March/139905.html.

Any idea when the rebase on LLVM 10 will happen?

andreisfr, 2020-03-22:
Hi @aykevl ! There is no significant problems to rebase Xtensa backend on LLVM 10, we already have version rebased on early LLVM 10.  I think soon after  LLVM 10.0.0 release we will verify last changes, test and publish new rebased version.

deadprogram, 2020-04-17:
LLVM 10 is now out. Will there be a rebased version soon? Thank you!

deadprogram, 2020-05-26:
Any action here? @andreisfr we really want this! :smile_cat:

andreisfr, 2020-05-29:
@deadprogram , we are preparing and testing new version, hope it will be available soon)

aykevl, 2020-07-04:
@andreisfr any update? A recent LLVM version is required for TinyGo and forking this fork just for Xtensa support is difficult.

AshUK, 2020-07-14:
+1 we are keen to see this as we currently use clang 10 for static analysis on our projects

andreisfr, 2020-07-15:
Hello @aykevl , @AshUK ! We significally refactored series of  patches implementing the Xtensa backend and ported it on Clang/LLVM release 10. So, this process took a some time. Current version is under review, probably review process will be finished in 1-2 weeks and then we will upload new version.

igrr, 2020-07-30:
Thanks for the patience everyone. The patches based on release 10 are now available in https://github.com/espressif/llvm-project/pull/30.

closed 2020-07-30 by igrr