espressif/llvm-project #20: Floating point number regression

lexxvir, 2020-04-23:
Hi,

I have encountered into the regression in the current xtensa_release_9.0.1 branch.
Previously I used GCC based branch and it works fine.
Worth noting that I also changed the Rust version, so it may be related. 

To reproduce the issue I have prepared minimal version of Rust code:
```rust
#[no_mangle]
#[inline(never)]
fn test_float_cmp() {
    let mut f: f32 = 10.0;
    let mut p = Printf {};

    while f > 0.0 {
        // works fine
        // let _ = writeln!(p, "{:04x}", f.to_bits());

        // works bad
        let _ = writeln!(p, "{}", f);

        f -= 0.1;
    }
}
```

* "works bad": loop produces only **one** value `10.0` and exits
* "works fine": loop produces 100 values

Some comments for the code:
* "bad" case formats `f32` to human readable string
* "good" case formats `u32` to hexadecimal number
* `to_bits` converts `f32` to `u32` as is at bits representation
* `Printf` is thin wrapper around regular C's `printf:
```rust
struct Printf;
impl Write for Printf {
    fn write_str(&mut self, s: &str) -> Result {
        for c in s.as_bytes() {
            unsafe { printf("%c\0".as_ptr() as *const _, *c as i32) }
        }
        Ok(())
    }
}
```

I've attached archive with: Rust source code demo, disassembled code for "good" and "bad" cases, final binary and LLVM bitcode files:

[float-issue.zip](https://github.com/espressif/llvm-project/files/4522417/float-issue.zip)

P.S. Actually I use https://github.com/mabezdev/llvm-project fork, but now it contains only one [commit](https://github.com/MabezDev/llvm-project/commit/1ad7aa125beddfb126516ba60f26a9bcb8103e51) ahead of this branch.

andreisfr, 2020-12-04:
Hi, @lexxvir ! Could you check this example using new branch xtensa_release_11.0.0, please?

lexxvir, 2020-12-13:
Hi @andreisfr 

I've checked xtensa_release_11.0.0 branch and confirm that the issue is fixed! Thanks for the support!

closed 2020-12-13 by lexxvir

andreisfr, 2020-12-14:
Hi, @lexxvir , thank you very much for your help!