espressif/llvm-project #31: Linking issues (libLTO.so)

katyo, 2020-08-01:
I have an issues with linking LTO shared library:
```
[ 70%] Linking CXX shared library ../../lib/libLTO.so
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: CMakeFiles/LTO.dir/lto.cpp.o: in function `lto_initialize() [clone .part.0]':
lto.cpp:(.text._ZL14lto_initializev.part.0+0x66): undefined reference to `LLVMInitializeXtensaTargetInfo'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: lto.cpp:(.text._ZL14lto_initializev.part.0+0xbb): undefined reference to `LLVMInitializeXtensaTargetInfo'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: lto.cpp:(.text._ZL14lto_initializev.part.0+0x110): undefined reference to `LLVMInitializeXtensaTarget'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: lto.cpp:(.text._ZL14lto_initializev.part.0+0x165): undefined reference to `LLVMInitializeXtensaTargetMC'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: lto.cpp:(.text._ZL14lto_initializev.part.0+0x1b0): undefined reference to `LLVMInitializeXtensaAsmParser'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: lto.cpp:(.text._ZL14lto_initializev.part.0+0x205): undefined reference to `LLVMInitializeXtensaAsmPrinter'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: CMakeFiles/LTO.dir/LTODisassembler.cpp.o: in function `lto_initialize_disassembler':
LTODisassembler.cpp:(.text.lto_initialize_disassembler+0x55): undefined reference to `LLVMInitializeXtensaTargetInfo'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: LTODisassembler.cpp:(.text.lto_initialize_disassembler+0xaa): undefined reference to `LLVMInitializeXtensaTargetMC'
/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ld: LTODisassembler.cpp:(.text.lto_initialize_disassembler+0xf5): undefined reference to `LLVMInitializeXtensaAsmParser'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/lto/CMakeFiles/LTO.dir/build.make:100: lib/libLTO.so.10] Error 1
make[1]: *** [CMakeFiles/Makefile2:25161: tools/lto/CMakeFiles/LTO.dir/all] Error 2
```

andreisfr, 2020-08-01:
Hello @katyo,, could you attach build command sequence to reproduce the problem?

katyo, 2020-08-01:
I applied xtensa patches from this repo to llvmorg-10.0.1

Cmake flags is used to build llvm:
```
-DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF \
-DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF \
-DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON \
-DCMAKE_BUILD_TYPE=Release \
-DCMAKE_SKIP_BUILD_RPATH=ON \
-DCMAKE_INSTALL_LOCALEDIR=/nix/store/928a11zzsh21s740kwk8gmsyhv6gbp0j-llvm-10.0.1-lib/share/locale \
-DCMAKE_INSTALL_LIBEXECDIR=/nix/store/928a11zzsh21s740kwk8gmsyhv6gbp0j-llvm-10.0.1-lib/libexec \
-DCMAKE_INSTALL_LIBDIR=/nix/store/928a11zzsh21s740kwk8gmsyhv6gbp0j-llvm-10.0.1-lib/lib \
-DCMAKE_INSTALL_DOCDIR=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1/share/doc/ \
-DCMAKE_INSTALL_INFODIR=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1/share/info \
-DCMAKE_INSTALL_MANDIR=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1/share/man \
-DCMAKE_INSTALL_OLDINCLUDEDIR=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1/include \
-DCMAKE_INSTALL_INCLUDEDIR=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1/include \
-DCMAKE_INSTALL_SBINDIR=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1/sbin \
-DCMAKE_INSTALL_BINDIR=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1/bin \
-DCMAKE_INSTALL_NAME_DIR=/nix/store/928a11zzsh21s740kwk8gmsyhv6gbp0j-llvm-10.0.1-lib/lib \
-DCMAKE_POLICY_DEFAULT_CMP0025=NEW \
-DCMAKE_OSX_SYSROOT= \
-DCMAKE_OSX_ARCHITECTURES=x86_64 \
-DCMAKE_FIND_FRAMEWORK=last \
-DCMAKE_STRIP=/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/strip \
-DCMAKE_RANLIB=/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ranlib \
-DCMAKE_AR=/nix/store/1zf4cnaaidjajwb4gx4mnkqc5dypkcdy-binutils-2.31.1/bin/ar \
-DCMAKE_C_COMPILER=gcc \
-DCMAKE_CXX_COMPILER=g++ \
-DCMAKE_INSTALL_PREFIX=/nix/store/rblp2rp2b3fqm2jz2agk5mhq8yifrdii-llvm-10.0.1 \
-DCMAKE_BUILD_TYPE=Release \
-DLLVM_INSTALL_UTILS=ON \
-DLLVM_BUILD_TESTS=ON \
-DLLVM_ENABLE_FFI=ON \
-DLLVM_ENABLE_RTTI=ON \
-DLLVM_HOST_TRIPLE=x86_64-unknown-linux-gnu \
-DLLVM_DEFAULT_TARGET_TRIPLE=x86_64-unknown-linux-gnu \
-DLLVM_ENABLE_DUMP=ON \
-DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=Xtensa \
-DLLVM_LINK_LLVM_DYLIB=ON \
-DLLVM_BINUTILS_INCDIR=/nix/store/r9gxdlp3mra863zarr14qfd97zagm1s9-libbfd-2.31.1-dev/include
```

katyo, 2020-08-01:
`$ nm --defined-only -C libLLVM-10.so | grep -i xtensa`:
```
0000000000af4c80 t _GLOBAL__sub_I_XtensaTargetInfo.cpp
00000000035dbd40 T LLVMInitializeXtensaDisassembler
00000000035dbad0 t createXtensaDisassembler(llvm::Target const&, llvm::MCSubtargetInfo const&, llvm::MCContext&)
00000000035dbb30 t (anonymous namespace)::XtensaDisassembler::~XtensaDisassembler()
00000000035dbb20 t (anonymous namespace)::XtensaDisassembler::~XtensaDisassembler()
00000000035dbb20 t (anonymous namespace)::XtensaDisassembler::~XtensaDisassembler()
0000000005972e20 b llvm::TheXtensaTarget
00000000035df5a0 t (anonymous namespace)::XtensaDisassembler::getInstruction(llvm::MCInst&, unsigned long&, llvm::ArrayRef<unsigned char>, unsigned long, llvm::raw_ostream&) const
00000000056ac970 d typeinfo for (anonymous namespace)::XtensaDisassembler
0000000004c1aa20 r typeinfo name for (anonymous namespace)::XtensaDisassembler
00000000056ac988 d vtable for (anonymous namespace)::XtensaDisassembler
0000000003a30e00 r llvm::Intrinsic::getIntrinsicForGCCBuiltin(char const*, llvm::StringRef)::xtensaNames
```
Unfortunattely that symbols missing in `libLLVM.so`

katyo, 2020-08-01:
I found that __LLVM_EXTERNAL_VISIBILITY__ is missing:
https://github.com/espressif/llvm-project/blob/713f3bf47aea13c2a390332eb6d4db9ec853ef43/llvm/lib/Target/Xtensa/TargetInfo/XtensaTargetInfo.cpp#L17
https://github.com/espressif/llvm-project/blob/713f3bf47aea13c2a390332eb6d4db9ec853ef43/llvm/lib/Target/ARM/TargetInfo/ARMTargetInfo.cpp#L30

closed 2020-08-03 by espressif-bot

andreisfr, 2020-08-03:
@katyo , thank you very much for your help! We corrected code and updated default branch.

commit 1659ba15 (2020-08-03): [Xtensa] Correct visibility of the Xtensa backend initialize functions.

Make backend initialize functions public and visible outside library.

Closes #31