espressif/llvm-project #33: Crash in floating point comparison

aykevl, 2020-08-05:
With the new LLVM 10 branch (at 1659ba158c4af4aacf1d331c33ebb7274c55386d) I got a crash when compiling the following code:

```llvm
target datalayout = "e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32"
target triple = "xtensa"

define void @__divsc3(float %x) {
entry:
  %cmp = fcmp ueq float %x, 0x7FF0000000000000
  br i1 %cmp, label %if.else, label %if.then

if.then:
  unreachable

if.else:
  unreachable
}
```

The crash is:

```
~/src/github.com/tinygo-org/tinygo$ ./llvm-build.xtensa/bin/llc bugpoint-reduced-simplified.ll
llc: /home/ayke/src/github.com/tinygo-org/tinygo/llvm-project.xtensa/llvm/lib/CodeGen/SelectionDAG/LegalizeDAG.cpp:978: void (anonymous namespace)::SelectionDAGLegalize::LegalizeOp(llvm::SDNode *): Assertion `(TLI.getTypeAction(*DAG.getContext(), Op.getValueType()) == TargetLowering::TypeLegal || Op.getOpcode() == ISD::TargetConstant || Op.getOpcode() == ISD::Register) && "Unexpected illegal type!"' failed.
Stack dump:
0.	Program arguments: ./llvm-build.xtensa/bin/llc bugpoint-reduced-simplified.ll 
1.	Running pass 'Function Pass Manager' on module 'bugpoint-reduced-simplified.ll'.
2.	Running pass 'Xtensa DAG->DAG Pattern Instruction Selection' on function '@__divsc3'
 #0 0x0000000002aa5044 PrintStackTraceSignalHandler(void*) (./llvm-build.xtensa/bin/llc+0x2aa5044)
 #1 0x0000000002aa2cde llvm::sys::RunSignalHandlers() (./llvm-build.xtensa/bin/llc+0x2aa2cde)
 #2 0x0000000002aa5455 SignalHandler(int) (./llvm-build.xtensa/bin/llc+0x2aa5455)
 #3 0x00007f757db7a730 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x12730)
 #4 0x00007f757d6ac7bb raise (/lib/x86_64-linux-gnu/libc.so.6+0x377bb)
 #5 0x00007f757d697535 abort (/lib/x86_64-linux-gnu/libc.so.6+0x22535)
 #6 0x00007f757d69740f (/lib/x86_64-linux-gnu/libc.so.6+0x2240f)
 #7 0x00007f757d6a5102 (/lib/x86_64-linux-gnu/libc.so.6+0x30102)
 #8 0x0000000002808c4a (anonymous namespace)::SelectionDAGLegalize::LegalizeOp(llvm::SDNode*) (./llvm-build.xtensa/bin/llc+0x2808c4a)
 #9 0x000000000280899a llvm::SelectionDAG::Legalize() (./llvm-build.xtensa/bin/llc+0x280899a)
#10 0x0000000002905ebd llvm::SelectionDAGISel::CodeGenAndEmitDAG() (./llvm-build.xtensa/bin/llc+0x2905ebd)
#11 0x0000000002903fbf llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) (./llvm-build.xtensa/bin/llc+0x2903fbf)
#12 0x0000000002900944 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) (./llvm-build.xtensa/bin/llc+0x2900944)
#13 0x0000000001fc04be llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (./llvm-build.xtensa/bin/llc+0x1fc04be)
#14 0x000000000238972c llvm::FPPassManager::runOnFunction(llvm::Function&) (./llvm-build.xtensa/bin/llc+0x238972c)
#15 0x0000000002389a08 llvm::FPPassManager::runOnModule(llvm::Module&) (./llvm-build.xtensa/bin/llc+0x2389a08)
#16 0x000000000238a110 llvm::legacy::PassManagerImpl::run(llvm::Module&) (./llvm-build.xtensa/bin/llc+0x238a110)
#17 0x000000000150fbc4 compileModule(char**, llvm::LLVMContext&) (./llvm-build.xtensa/bin/llc+0x150fbc4)
#18 0x000000000150d6fd main (./llvm-build.xtensa/bin/llc+0x150d6fd)
#19 0x00007f757d69909b __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2409b)
#20 0x000000000150b02a _start (./llvm-build.xtensa/bin/llc+0x150b02a)
```

This is a reduced version of a panic when trying to compile the file [divsc3.c of compiler-rt](https://github.com/llvm-mirror/compiler-rt/blob/5bc79797e1f9184f39e1bd30fac75d995b72cea3/lib/builtins/divsc3.c).

andreisfr, 2020-08-18:
Hi @aykevl , could I ask you to describe more precisely, how to generate such llvm IR file, to reproduce the problem? It seems that in __divsc3 function argument must be i32 value instead of float. Thank you.

aykevl, 2020-08-30:
The name of the function is not relevant, this works as well:

```llvm
target datalayout = "e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32"
target triple = "xtensa"

define void @foo(float %x) {
entry:
  %cmp = fcmp ueq float %x, 0x7FF0000000000000
  br i1 %cmp, label %if.else, label %if.then

if.then:
  unreachable

if.else:
  unreachable
}
```

You can easily reproduce it with `llc`, such as with

    ./llvm-build.xtensa/bin/llc crash.ll

I can try to come up with a reproducer in C. I made it in LLVM IR as that's usually easier to debug (I found this simplified code sample using `bugpoint`, but it is originally based on [divsc3.c](https://github.com/llvm-mirror/compiler-rt/blob/5bc79797e1f9184f39e1bd30fac75d995b72cea3/lib/builtins/divsc3.c) in compiler-rt).

aykevl, 2020-08-30:
> It seems that in __divsc3 function argument must be i32 value instead of float.

Ah maybe I understand the confusion. Are you wondering about the big `0x7FF0000000000000` constant? It looks like an integer but it is actually a raw floating point constant. I guess LLVM stores it in hexadecimal format to avoid ambiguity on the bit pattern.

aykevl, 2020-08-30:
I managed to produce a C file which crashes in the same way:

```c
__attribute__((noreturn))
void func1(void);
__attribute__((noreturn))
void func2(void);

void foo(float x) {
  if (x == __builtin_huge_valf()) {
    func1();
  } else {
    func2();
  }
}
```

Compile with:

    clang --target=xtensa -Os -c -o xtensa-crash.o xtensa-crash.c

commit 348e54d4 (2020-10-15): [Xtensa] Correct lowering BR_CC with FP operands.

Remove register class for boolean operands, because it is only suitable for FP compare operations
and may lead to problems in other cases. Disable load width reduction, because for IRAM memory
it may cause exceptions.
Closes #33, #38.
closed 2020-10-15 by espressif-bot