espressif/llvm-project #34: Frontend crash with picolibc

aykevl, 2020-08-05:
When compiling picolibc (for use with TinyGo), I get a frontend crash. Because this crash only happens when targeting the Xtensa backend (and not with any of the other backends such as AVR, ARM, RISC-V) I assume it's somehow a problem with the Xtensa support in Clang.

[strdup.zip](https://github.com/espressif/llvm-project/files/5031422/strdup.zip)

You can compile the C file in the zip file with the following command:

    llvm-build.xtensa/bin/clang-10 --target=xtensa -emit-llvm -S -nostdlibinc strdup-501ff8.c

The crash is the following:

```
Wrong types for attribute: inalloca nest noalias nocapture nonnull readnone readonly sret byval dereferenceable(1) dereferenceable_or_null(1)
i32 (i32)* @strdup
Wrong types for attribute: inalloca nest noalias nocapture nonnull readnone readonly sret byval dereferenceable(1) dereferenceable_or_null(1)
  %call2 = call noalias i32 @malloc(i32 %2) #4
in function strdup
fatal error: error in backend: Broken function found, compilation aborted!
Stack dump:
0.	Program arguments: /home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10 --target=xtensa -emit-llvm -S -nostdlibinc strdup-501ff8.c 
1.	<eof> parser at end of file
2.	Per-function optimization
3.	Running pass 'Module Verifier' on function '@strdup'
 #0 0x0000000003694834 PrintStackTraceSignalHandler(void*) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3694834)
 #1 0x00000000036924ce llvm::sys::RunSignalHandlers() (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x36924ce)
 #2 0x00000000036939dd llvm::sys::CleanupOnSignal(unsigned long) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x36939dd)
 #3 0x000000000361d563 (anonymous namespace)::CrashRecoveryContextImpl::HandleCrash(int, unsigned long) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x361d563)
 #4 0x000000000361d4df llvm::CrashRecoveryContext::HandleExit(int) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x361d4df)
 #5 0x000000000368f0ae (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x368f0ae)
 #6 0x000000000222a5c2 (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x222a5c2)
 #7 0x0000000003622a74 llvm::report_fatal_error(llvm::Twine const&, bool) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3622a74)
 #8 0x000000000362296f (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x362296f)
 #9 0x00000000030d222f (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x30d222f)
#10 0x000000000305ea4c llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x305ea4c)
#11 0x000000000305e0a9 llvm::legacy::FunctionPassManagerImpl::run(llvm::Function&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x305e0a9)
#12 0x000000000305df3b llvm::legacy::FunctionPassManager::run(llvm::Function&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x305df3b)
#13 0x000000000386a945 clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::HeaderSearchOptions const&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::DataLayout const&, llvm::Module*, clang::BackendAction, std::unique_ptr<llvm::raw_pwrite_stream, std::default_delete<llvm::raw_pwrite_stream> >) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x386a945)
#14 0x00000000040390b5 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x40390b5)
#15 0x0000000004debc73 clang::ParseAST(clang::Sema&, bool, bool) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x4debc73)
#16 0x0000000003f86bb0 clang::FrontendAction::Execute() (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3f86bb0)
#17 0x0000000003ee3343 clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3ee3343)
#18 0x0000000004032980 clang::ExecuteCompilerInvocation(clang::CompilerInstance*) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x4032980)
#19 0x000000000222a1dd cc1_main(llvm::ArrayRef<char const*>, char const*, void*) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x222a1dd)
#20 0x00000000022281ec ExecuteCC1Tool(llvm::SmallVectorImpl<char const*>&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x22281ec)
#21 0x0000000003dab942 void llvm::function_ref<void ()>::callback_fn<clang::driver::CC1Command::Execute(llvm::ArrayRef<llvm::Optional<llvm::StringRef> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, bool*) const::$_1>(long) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3dab942)
#22 0x000000000361d481 llvm::CrashRecoveryContext::RunSafely(llvm::function_ref<void ()>) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x361d481)
#23 0x0000000003daae1d clang::driver::CC1Command::Execute(llvm::ArrayRef<llvm::Optional<llvm::StringRef> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, bool*) const (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3daae1d)
#24 0x0000000003d787bb clang::driver::Compilation::ExecuteCommand(clang::driver::Command const&, clang::driver::Command const*&) const (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3d787bb)
#25 0x0000000003d78baa clang::driver::Compilation::ExecuteJobs(clang::driver::JobList const&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*> >&) const (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3d78baa)
#26 0x0000000003d8fad8 clang::driver::Driver::ExecuteCompilation(clang::driver::Compilation&, llvm::SmallVectorImpl<std::pair<int, clang::driver::Command const*> >&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x3d8fad8)
#27 0x0000000002227bc3 main (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x2227bc3)
#28 0x00007ff131c3009b __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2409b)
#29 0x000000000222502a _start (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build.xtensa/bin/clang-10+0x222502a)
```

aykevl, 2020-08-31:
I have reduced this crash to the following:

```c
typedef __SIZE_TYPE__ size_t;
void *malloc(size_t) __attribute__((__malloc__));
void foobar(void) {
  malloc(3);
}
```

Compile using:
```
clang --target=xtensa -c -o xtensa-malloc-crash.o xtensa-malloc-crash.c
```

The stack trace is as above.

This issue is preventing picolibc from being compiled for Xtensa, which is important for TinyGo.

aykevl, 2020-09-05:
I've gotten a bit closer. If you take this code:

```c
void foobar(void *x) {
}
```

It is compiled to this IR:

```llvm
; Function Attrs: norecurse nounwind optsize readnone
define dso_local void @foobar(i32 %x.coerce) local_unnamed_addr #0 {
entry:
  ret void
}
```

As you can see, `void*` is lowered to `i32`. That doesn't make a lot of sense to me, and is probably why malloc doesn't work.

aykevl, 2020-09-05:
I have a possible fix in #37, it at least gets the file to compile.

andreisfr, 2020-09-25:
Hi @aykevl , thank you very much for investigation of the problem! We will use these results to correct the problem.

commit 7cc7ee08 (2020-10-15): [Xtensa] Correct Call ABI for function return arguments.

Closes #34
closed 2020-10-15 by espressif-bot