espressif/llvm-project #35: clang fails to compile esp-idf

AshUK, 2020-08-13:
I'm attempting to build to esp-idf 4.1 using clang from this repo. If this is not currently possible or with this version of the IDF please let me know.

The clang compiler spews the following error:

```
/home/hello/esp-idf/components/bt/controller/bt.c:756:5: error: invalid register name
    XTHAL_SET_INTSET((1<<intr_no));
    ^
/home/hello/esp-idf/components/xtensa/include/xtensa/core-macros.h:341:25: note: expanded from macro 'XTHAL_SET_INTSET'
                        __asm__ __volatile__("wsr.intset %0" :: "a"(__interrupt):"memory"); \
                                             ^
<inline asm>:1:2: note: instantiated into assembly here
        wsr.intset a8
```

andreisfr, 2020-09-25:
Please, try to use experimental prebuild clang toolchain for linux https://dl.espressif.com/dl/xtensa-esp32-elf-llvm10_0_1-esp-10.0.1-20200903-linux-amd64.tar.xz. To compile ESP-IDF projects you need to change C/C++ compilers in ESP-IDF\tools\cmake\toolchain-esp32.cmake to xtensa-esp32-elf-clang(clang++).

AshUK, 2020-11-13:
@andreisfr Thank you for sharing the build this has saved me a lot of time already. However it does still fail to build both the release/v4.1 and current master versions of the IDF. Do you know of any specific commit that you know works ?

The good news is I can compile my own specific source files.

Also do you think you will be to provide a version clang-tidy/sa that uses this llvm backend ?

Version:
```
hello@office-srv:~/esp-blinky$ xtensa-esp32-elf-clang --version
clang version 10.0.1 (https://gitlab-ci-token:qEDMeHsN_HUZVvpsR9rL@gitlab.espressif.cn:6688/idf/llvm-project.git ca636bf4c294e2339ad15f81e0e9359dfb49b254)
Target: xtensa
Thread model: posix
InstalledDir: /home/hello/.espressif/tools/xtensa-esp32-elf-clang/bin
```

Commit espressif/esp-idf@494a124d961c4c755685b22fe986c31826d6f503:
```
hello@office-srv:~/esp-blinky$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/hello/esp-blinky/build
Executing "ninja all"...
[1/630] Performing build step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-build bootloader/bootloader.elf bootloader/bootloader.bin bootloader/bootloader.map
cd /home/hello/esp-blinky/build/bootloader && /usr/bin/cmake --build .
[1/11] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_sleep.c.obj
/home/hello/esp-idf/components/esp_hw_support/port/esp32/rtc_sleep.c:282:19: error: unexpected token
                 "bbci a2, "STR(RTC_MEM_CRC_FINISH_S)", .Lwaitcrc\n"
                  ^
<inline asm>:8:13: note: instantiated into assembly here
bbci a2, (31), .Lwaitcrc
            ^
1 error generated.
[2/11] Linking C static library esp-idf/esp_hw_support/libesp_hw_support.a
FAILED: esp-idf/esp_hw_support/libesp_hw_support.a
: && /usr/bin/cmake -E remove esp-idf/esp_hw_support/libesp_hw_support.a && /home/hello/.espressif/tools/xtensa-esp32-elf-clang/bin/xtensa-esp32-elf-ar qc esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/compare_set.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/cpu_util.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_clk.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_clk_init.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_init.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_pm.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_sleep.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_time.c.obj esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_wdt.c.obj && /home/hello/.espressif/tools/xtensa-esp32-elf-clang/bin/xtensa-esp32-elf-ranlib esp-idf/esp_hw_support/libesp_hw_support.a && :
/home/hello/.espressif/tools/xtensa-esp32-elf-clang/bin/xtensa-esp32-elf-ar: esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_sleep.c.obj: No such file or directory
ninja: build stopped: subcommand failed.
[2/630] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_sleep.c.obj
/home/hello/esp-idf/components/esp_hw_support/port/esp32/rtc_sleep.c:282:19: error: unexpected token
                 "bbci a2, "STR(RTC_MEM_CRC_FINISH_S)", .Lwaitcrc\n"
                  ^
<inline asm>:8:13: note: instantiated into assembly here
bbci a2, (31), .Lwaitcrc
            ^
1 error generated.
[3/630] Building CXX object esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_guards.cpp.obj
/home/hello/esp-idf/components/cxx/cxx_guards.cpp:175:17: error: '__cxa_guard_release' is missing exception specification 'throw()'
extern "C" void __cxa_guard_release(__guard* pg)
                ^
                                                 throw()
/home/hello/esp-idf/components/cxx/cxx_guards.cpp:37:17: note: previous declaration is here
extern "C" void __cxa_guard_release(__guard* pg);
                ^
/home/hello/esp-idf/components/cxx/cxx_guards.cpp:195:17: error: '__cxa_guard_abort' is missing exception specification 'throw()'
extern "C" void __cxa_guard_abort(__guard* pg)
                ^
                                               throw()
/home/hello/esp-idf/components/cxx/cxx_guards.cpp:38:17: note: previous declaration is here
extern "C" void __cxa_guard_abort(__guard* pg);
                ^
2 errors generated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

gerekon, 2022-12-05:
Closing as reported problem (and some otherts) were fixed in espressif/esp-idf@47c2c13de5103ce022a0987f4a8b45b85f824040

closed 2022-12-05 by gerekon