espressif/llvm-project #36: publish LVVM builds

AshUK, 2020-08-13:
Currently it takes over an hour to build LVVM on a Quad Core i7. It takes even longer within a github virtual environment as demonstrated by github actions on this fork. It would be ideal if you could build binaries as releases on this project. Specifically clang, and clang extra tools (clang-format, clang-tidy). We currently use this toolset on our internal CI builds but with main lvvm project installed with apt-get, this solution is not ideal as we have to setup complex configuration in order to successfully analyse our build.

We are interested in using the clang-tidy and the clang static analysis tools to improve code quality and prevent bugs.

andreisfr, 2020-09-25:
Hi @AshUK, you could try to use experimental prebuild clang toolchain for linux https://dl.espressif.com/dl/xtensa-esp32-elf-llvm10_0_1-esp-10.0.1-20200903-linux-amd64.tar.xz. To compile ESP-IDF projects you need to change C/C++ compilers in ESP-IDF\tools\cmake\toolchain-esp32.cmake to xtensa-esp32-elf-clang(clang++).

igrr, 2021-12-03:
We are now publishing release binaries at https://github.com/espressif/llvm-project/releases. New release based on LLVM 13 will be made soon.

closed 2021-12-03 by igrr