espressif/llvm-project #38: Comparing a single bit in a value leads to a crash

aykevl, 2020-09-05:
I found an interesting bug in the Xtensa backend, which I'm hitting in newlib/libc/string/strverscmp.c of picolibc (which is based on newlib):

```
llc: /home/ayke/src/github.com/tinygo-org/tinygo/llvm-project.xtensa/llvm/lib/CodeGen/SelectionDAG/SelectionDAG.cpp:8357: void llvm::SelectionDAG::ReplaceAllUsesWith(llvm::SDNode *, llvm::SDNode *): Assertion `(!From->hasAnyUseOfValue(i) || From->getValueType(i) == To->getValueType(i)) && "Cannot use this version of ReplaceAllUsesWith!"' failed.
Stack dump:
0.	Program arguments: ./llvm-build.xtensa/bin/llc bit-is-set.ll 
1.	Running pass 'Function Pass Manager' on module 'bit-is-set.ll'.
2.	Running pass 'Xtensa DAG->DAG Pattern Instruction Selection' on function '@bit_is_set'
 #0 0x0000000002aa5044 PrintStackTraceSignalHandler(void*) (./llvm-build.xtensa/bin/llc+0x2aa5044)
 #1 0x0000000002aa2cde llvm::sys::RunSignalHandlers() (./llvm-build.xtensa/bin/llc+0x2aa2cde)
 #2 0x0000000002aa5455 SignalHandler(int) (./llvm-build.xtensa/bin/llc+0x2aa5455)
 #3 0x00007f3b43cfa730 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x12730)
 #4 0x00007f3b4382c7bb raise (/lib/x86_64-linux-gnu/libc.so.6+0x377bb)
 #5 0x00007f3b43817535 abort (/lib/x86_64-linux-gnu/libc.so.6+0x22535)
 #6 0x00007f3b4381740f (/lib/x86_64-linux-gnu/libc.so.6+0x2240f)
 #7 0x00007f3b43825102 (/lib/x86_64-linux-gnu/libc.so.6+0x30102)
 #8 0x00000000028b2317 llvm::SelectionDAG::ReplaceAllUsesWith(llvm::SDNode*, llvm::SDNode*) (./llvm-build.xtensa/bin/llc+0x28b2317)
 #9 0x0000000002767c71 llvm::SelectionDAG::Combine(llvm::CombineLevel, llvm::AAResults*, llvm::CodeGenOpt::Level) (./llvm-build.xtensa/bin/llc+0x2767c71)
#10 0x000000000290506d llvm::SelectionDAGISel::CodeGenAndEmitDAG() (./llvm-build.xtensa/bin/llc+0x290506d)
#11 0x0000000002903fbf llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) (./llvm-build.xtensa/bin/llc+0x2903fbf)
#12 0x0000000002900944 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) (./llvm-build.xtensa/bin/llc+0x2900944)
#13 0x0000000001fc04be llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (./llvm-build.xtensa/bin/llc+0x1fc04be)
#14 0x000000000238972c llvm::FPPassManager::runOnFunction(llvm::Function&) (./llvm-build.xtensa/bin/llc+0x238972c)
#15 0x0000000002389a08 llvm::FPPassManager::runOnModule(llvm::Module&) (./llvm-build.xtensa/bin/llc+0x2389a08)
#16 0x000000000238a110 llvm::legacy::PassManagerImpl::run(llvm::Module&) (./llvm-build.xtensa/bin/llc+0x238a110)
#17 0x000000000150fbc4 compileModule(char**, llvm::LLVMContext&) (./llvm-build.xtensa/bin/llc+0x150fbc4)
#18 0x000000000150d6fd main (./llvm-build.xtensa/bin/llc+0x150d6fd)
#19 0x00007f3b4381909b __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2409b)
#20 0x000000000150b02a _start (./llvm-build.xtensa/bin/llc+0x150b02a)
```

The LLVM IR is as follows:

```llvm
target datalayout = "e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32"
target triple = "xtensa"

define i8 @bit_is_set(i8 %a) {
  %b = and i8 %a, 16
  %bool = icmp eq i8 %b, 0
  br i1 %bool, label %true, label %false

true:
  ret i8 1

false:
  ret i8 0
}
```

Compile this with:

    llc bit-is-set.ll

I also managed to come up with some C code that crashes in the same way:

```c
void noop(char);

void bit_is_set(char a) {
	char b = a & 16;
	if (b == 0) {
		noop(3);
		__builtin_unreachable();
	} else {
		noop(0);
		__builtin_unreachable();
	}
}
```

Compile using:

    clang --target=xtensa -c -Os -o bit-is-set.o bit-is-set.c

It appears that it is triggered when checking for a single bit in an integer variable and branching based on that. As it happens, I recently looked at the Xtensa architecture manual and saw that there is a branch instruction that branches based on whether a single bit is set in a register, so I'm guessing the crash is related to that instruction.

andreisfr, 2020-10-15:
Hi @aykevl ! Please try to use xtensa_release_10.0.1 branch with latest commits

gerekon, 2020-10-15:
Cosed by 348e54d

closed 2020-10-15 by gerekon