espressif/llvm-project #40: LLVM 11 support

aykevl, 2020-11-05:
LLVM 11 was released a few weeks ago. Any idea when this Xtensa fork will be updated to LLVM 11?

katyo, 2020-11-05:
> LLVM 11 was released a few weeks ago. Any idea when this Xtensa fork will be updated to LLVM 11?

Despite of which is not so hard, but where is any reasons to prioritize rebase to latest llvm when more important things still undone.

aykevl, 2020-11-06:
> > LLVM 11 was released a few weeks ago. Any idea when this Xtensa fork will be updated to LLVM 11?
> 
> Despite of which is not so hard, but where is any reasons to prioritize rebase to latest llvm when more important things still undone.

It is important because otherwise [TinyGo](https://tinygo.org/) might drop support for the ESP32 and ESP8266. We need LLVM 11 for improvements (especially in the AVR backend) so staying on LLVM 10 is not an option.

andreisfr, 2020-11-10:
Hi @aykevl ! We rebased Xtensa backend on LLVM 11, now we verifying/debugging it, probably it will be ready for upload to github in 1-2 weeks.

igrr, 2020-11-30:
@aykevl the branch based on release/11.x has been published: https://github.com/espressif/llvm-project/pull/42.

aykevl, 2020-12-03:
Thanks a lot! Then I consider this issue fixed.

closed 2020-12-03 by aykevl