espressif/llvm-project #41: Floating point comparison miscompilation

aykevl, 2020-11-12:
I have the following C code:

```c
void sideeffect(void);

int identify(float v) {
    if (v+v == v && v > 0) {
        return 2;
    }
    if (v < 0) {
        sideeffect();
        return 5;
    }
    return 4;
}
```

When I call this with `identify(20)` (where 20 is the floating point value with bits 0x41a00000, see [here](https://play.golang.org/p/PN4Wg2zZsmv)) the function will return 5. This is clearly incorrect, as `v` is 20 which is bigger than 0.

I found the issue originally in TinyGo (runtime.printfloat32) but managed to reduce it and convert it to C while still maintaining the incorrect behavior.

andreisfr, 2020-11-30:
Hi @aykevl ! This problem corrected in xtensa_release_11.0.0 branch, please check it

aykevl, 2020-12-03:
Thank you! I have tested this and it is indeed fixed with the LLVM 11 branch.

closed 2020-12-03 by aykevl