espressif/llvm-project #43: Not supported instr: <MCInst 237 <MCOperand Reg:42>>

MabezDev, 2021-01-17:
I've just been updating the Xtensa Rust fork and I've ran into this issue when building for the esp32 in release mode (haven't yet reproduced on the esp8266).

```rust
Compiling core v0.0.0 (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/src/rust/library/core)
LLVM ERROR: Not supported instr: <MCInst 237 <MCOperand Reg:42>>
error: could not compile `core`
``` 

Here's the llvm IR: [core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll.zip](https://github.com/espressif/llvm-project/files/5827033/core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll.zip)

kelnos, 2021-01-21:
Ran into this as well with your branch.  Tried a few things to see if I could figure out what was going on, with no success:

1. Merged branch `rustc/11.0-2020-10-12` of  `rust-lang/llvm-project` (branch that 1.49.0 uses) into `xtensa_release_11.0.0`
2. Merged upstream `llvm/llvm-project`'s 11.x branch into `xtensa_release_11.0.0`
3. Attempted to bisect rust itself between 1.48.0 and 1.49.0

The bisect was unfortunately unhelpful, as rust's history between versions isn't linear; the 1.48.0 and 1.49.0 tags don't have much in common wrt commit history, so the bisect wasn't able to come up with anything conclusive.  Best it could do ended with:
```
The merge base 6ebad43c255e63ac0734646fe817de5780d76b45 is bad.
This means the bug has been fixed between 6ebad43c255e63ac0734646fe817de5780d76b45 and [7eac88abb2e57e752f3302f02be5f3ce3d7adfb4].
bisect run failed:
'bisect_state bad' exited with error code 3
```
Here's the (somewhat janky) script I used with `git bisect run`, in case it's useful: [test-rust-xtensa-bisect.sh.txt](https://github.com/espressif/llvm-project/files/5852221/test-rust-xtensa-bisect.sh.txt).  There are some paths in there you might need to edit if you want to try it yourself.

MabezDev, 2021-01-24:
I've done A little more analysis of the llvm ir.

The issue arises when optimizations are enabled.

```bash
# works
llc -filetype=obj core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll -O0

# fails
llc -filetype=obj core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll -O0
```

Fortunately, that last command tells  us some more info, namely that the offending function is `@_ZN4core3num6bignum5tests6Big8x310mul_digits17hbefc9cd21174a3deE`:

```bash
LLVM ERROR: Not supported instr: <MCInst 237 <MCOperand Reg:42>>
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: /home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc -filetype=obj core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll -O1 
1.      Running pass 'Function Pass Manager' on module 'core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll'.
2.      Running pass 'Xtensa Assembly Printer' on function '@_ZN4core3num6bignum5tests6Big8x310mul_digits17hbefc9cd21174a3deE'
 #0 0x0000556c3886ff3a llvm::sys::PrintStackTrace(llvm::raw_ostream&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x2243f3a)
 #1 0x0000556c3886de04 llvm::sys::RunSignalHandlers() (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x2241e04)
 #2 0x0000556c3886df48 SignalHandler(int) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x2241f48)
 #3 0x00007fc863e3cc10 __restore_rt (/lib64/libpthread.so.0+0x12c10)
 #4 0x00007fc8638a4de1 raise (/lib64/libc.so.6+0x38de1)
 #5 0x00007fc86388e536 abort (/lib64/libc.so.6+0x22536)
 #6 0x0000556c387e9756 llvm::report_fatal_error(llvm::Twine const&, bool) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x21bd756)
 #7 0x0000556c387e98ae (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x21bd8ae)
 #8 0x0000556c37963105 (anonymous namespace)::XtensaMCCodeEmitter::encodeInstruction(llvm::MCInst const&, llvm::raw_ostream&, llvm::SmallVectorImpl<llvm::MCFixup>&, llvm::MCSubtargetInfo const&) const (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1337105)
 #9 0x0000556c381efd65 llvm::MCELFStreamer::emitInstToData(llvm::MCInst const&, llvm::MCSubtargetInfo const&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1bc3d65)
#10 0x0000556c382050da llvm::MCObjectStreamer::emitInstructionImpl(llvm::MCInst const&, llvm::MCSubtargetInfo const&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1bd90da)
#11 0x0000556c382052b3 llvm::MCObjectStreamer::emitInstruction(llvm::MCInst const&, llvm::MCSubtargetInfo const&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1bd92b3)
#12 0x0000556c377afce5 llvm::XtensaAsmPrinter::emitInstruction(llvm::MachineInstr const*) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1183ce5)
#13 0x0000556c37b77871 llvm::AsmPrinter::emitFunctionBody() (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x154b871)
#14 0x0000556c37413795 llvm::AsmPrinter::runOnMachineFunction(llvm::MachineFunction&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0xde7795)
#15 0x0000556c37d76e14 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x174ae14)
#16 0x0000556c381288c0 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1afc8c0)
#17 0x0000556c38129e91 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1afde91)
#18 0x0000556c38127b08 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1afbb08)
#19 0x0000556c36c0ab8b compileModule(char**, llvm::LLVMContext&) (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x5deb8b)
#20 0x0000556c36b7ff6e main (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x553f6e)
#21 0x00007fc86388fe6b __libc_start_main (/lib64/libc.so.6+0x23e6b)
#22 0x0000556c36c0434a _start (/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x5d834a)
[1]    30229 abort       -filetype=obj core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll -O1
```
Here's the ASM for the offending function, I didn't immediately spot anything out of place:

```asm
# -- Begin function _ZN4core3num6bignum5tests6Big8x310mul_digits17hbefc9cd21174a3deE
	.literal .LCPI292_0, .Lalloc23096
	.literal .LCPI292_1, _ZN4core9panicking18panic_bounds_check17h5d2bbbfe824db14cE
	.literal .LCPI292_2, 65280
	.literal .LCPI292_3, _ZN4core5slice5index24slice_end_index_len_fail17h7f5b2f0bae923e3cE
	.global	_ZN4core3num6bignum5tests6Big8x310mul_digits17hbefc9cd21174a3deE
	.p2align	2
	.type	_ZN4core3num6bignum5tests6Big8x310mul_digits17hbefc9cd21174a3deE,@function
[... cut ...]
.Lfunc_end292:
	.size	_ZN4core3num6bignum5tests6Big8x310mul_digits17hbefc9cd21174a3deE, .Lfunc_end292-_ZN4core3num6bignum5tests6Big8x310mul_digits17hbefc9cd21174a3deE
                                        # -- End function
```

Here's the full asm, created by running `llc -filetype=asm core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.ll -O1`.

[core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.s.zip](https://github.com/espressif/llvm-project/files/5861391/core-e40c46572e76fe97.core.7bmkbcle-cgu.0.rcgu.s.zip)

kelnos, 2021-01-24:
Ah interesting.  In my efforts to look into the new work you started, I ended up building a debug version of llvm/rust, and with debug assertions enabled, I get this now:

```
argo xbuild --example esp32 --features="xtensa-lx-rt/lx6,xtensa-lx/lx6,esp32-hal"
   Compiling core v0.0.0 (/home/brian/src/esp32-rust/rust-xtensa/library/core)
   Compiling compiler_builtins v0.1.36
   Compiling rustc-std-workspace-core v1.99.0 (/home/brian/src/esp32-rust/rust-xtensa/library/rustc-std-workspace-core)
rustc: /home/brian/src/esp32-rust/rust-xtensa/src/llvm-project/llvm/lib/CodeGen/BranchRelaxation.cpp:335: bool {anonymous}::BranchRelaxation::fixupConditionalBranch(llvm::MachineInstr&): Assertion `!Fail && "branches to be relaxed must be analyzable"' failed.
error: could not compile `core`

Caused by:
  process didn't exit successfully: `/home/brian/src/esp32-rust/rust-xtensa/build/x86_64-unknown-linux-gnu/stage2/bin/rustc --crate-name core --edition=2018 /home/brian/src/esp32-rust/rust-xtensa/library/core/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=eefe0737f1bad7ee -C extra-filename=-eefe0737f1bad7ee --out-dir /tmp/cargo-xbuildIMe6T8/target/xtensa-esp32-none-elf/release/deps --target xtensa-esp32-none-elf -L dependency=/tmp/cargo-xbuildIMe6T8/target/xtensa-esp32-none-elf/release/deps -L dependency=/tmp/cargo-xbuildIMe6T8/target/release/deps -Cembed-bitcode=yes` (signal: 6, SIGABRT: process abort signal)
warning: build failed, waiting for other jobs to finish...
error: build failed
error: `"/home/brian/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/cargo-xbuildIMe6T8/Cargo.toml" "--target" "xtensa-esp32-none-elf" "--" "-Z" "force-unstable-if-unmarked"` failed with exit code: Some(101)
```

`BranchRelaxation::fixupConditionalBranch()` does sound like something related to the optimizer, so that makes sense that you found that it works ok with optimizations turned off.

Interestingly, if I look at that code, the result that triggers the assertion failure is silently ignored when assertions are disabled.

kelnos, 2021-01-28:
Figured out the offending commit in LLVM on the `xtensa_release_11.0.0` branch: 22aa88325c8100c7dca6ff28f26ab6e24b43bac5

I built rust 1.47.0 against the tip of the `xtensa_release_11.0.0` branch, and noticed the same LLVM error.  Made me realize that the `xtensa-target` branch of rust-xtensa was _not_ pointing to the tip of that branch, but to a few commits behind it.  Building the `xtensa-update-` branch against LLVM 9ecb19f774994a3efff5a6b89aa43ba2b8d2dd23 results in a seemingly-functioning rustc; at least rust-xtensa-quickstart now compiles properly.

stappersg, 2021-01-28:
>  results in a seemingly-functioning rustc; at least rust-xtensa-quickstart now compiles properly.

Please translate that into "issue should remain open (for ....)" or "close issue"

kelnos, 2021-01-28:
> Please translate that into "issue should remain open (for ....)" or "close issue"

Issue should remain open because the bug exists on the tip of the llvm `xtensa_release_11.0.0` branch.  Commit 22aa883 introduced this regression.

KerryRJ, 2021-01-28:
I've built a customized rust using 1.50 beta/1.51 nightly and the Espressif xtensa_release_11.0.0, and an updated esp32-hal and it is compiling the Quickstart, for Esp32, with no problems. I am building on Windows 10 using cargo and not Xargo or Xbuild.

MabezDev, 2021-01-28:
Great spot @kelnos! Awesome work, I've rolled the rust fork back to https://github.com/espressif/llvm-project/commit/9ecb19f774994a3efff5a6b89aa43ba2b8d2dd23 https://github.com/MabezDev/rust-xtensa/pull/44 and rebuilt and it works perfectly! Will merge to the `xtensa-target` branch soon.

> I've built a customized rust using 1.50 beta/1.51 nightly and the Espressif xtensa_release_11.0.0, and an updated esp32-hal and it is compiling the Quickstart, for Esp32, with no problems. I am building on Windows 10 using cargo and not Xargo or Xbuild.

Does it still work if you build in release? If you're using cargo via `-Z build-std` then it will compile `core` in debug too, where as Xargo etc will always compile core in release mode.

MabezDev, 2021-01-28:
@andreisfr FYI this is the rust code that causes the error: https://github.com/rust-lang/rust/blob/master/library/core/src/num/bignum.rs#L325
Unfortunately its wrapped up in a macro and only fails in [this instance](https://github.com/rust-lang/rust/blob/master/library/core/src/num/bignum.rs#L471).

I will try and produce a minimal C example if I have time over the weekend.

KerryRJ, 2021-01-29:
I was using `cargo build --example esp32 --release` to get the quickstart building with no core. The error does indeed appear when adding `-Z build-std`.

andreisfr, 2021-01-29:
> @andreisfr FYI this is the rust code that causes the error: https://github.com/rust-lang/rust/blob/master/library/core/src/num/bignum.rs#L325
> Unfortunately its wrapped up in a macro and only fails [...]

Hi @MabezDev, thank you very much for your investigation! It would be very helpful to have an С example which demonstrates the problem to solve it quickly.

Thanks also to all for support!

aykevl, 2021-02-03:
Here is a small reproducer in IR form:

```llvm
target datalayout = "e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32"
target triple = "xtensa-none-unknown-elf"

define void @crashing() {
entry:
  br i1 undef, label %bb1, label %bb2

bb1:
  %a = phi i32 [ 0, %entry ], [ %b, %bb2 ]
  unreachable

bb2:
  %b = phi i32 [ %c, %bb2 ], [ 1, %entry ]
  %c = add i32 %b, 1
  %cond = icmp eq i32 %b, 0
  br i1 %cond, label %bb1, label %bb2
}
```

aykevl, 2021-02-03:
I managed to get this small IR reproducer using `bugpoint`, `opt`, and some manual reducing of the function.

1. `bugpoint`: I ran `bugpoint --llc-safe xtensa-issue43.ll` to reduce the file as far as possible. Unfortunately, `bugpoint` was hanging during the "Attempting to reduce the number of function attributes in the testcase" but luckily it had produced a bugpoint-reduced-function.bc file just before that hang which I could work with. It's a large bitcode file (easily disassembled with `llvm-dis`) but where only one function still has a body: clearly the offending function. Still, I needed to reduce it further.
2. As a solution, I ran `opt -O2` on the resulting file, in an attempt to remove the function attributes `bugpoint` was hanging at.
3. This worked. A further `bugpoint --llc-safe` on the optimized file resulted in IR that was just 26 lines long - much easier to work with!
4. With some fine tuning I managed to bring down the file size further, to just 17 lines of IR (7 instructions over 3 basic blocks).

Normally only `bugpoint` with some manual tuning is necessary, but in this case clearly some creativity was needed to reduce this bug. I hope this helps people interested in reducing test cases for LLVM crashes.

andreisfr, 2021-02-09:
Hi, @aykevl! Thank you very much for example. Did you observed similar llc's output in example?

```
Assertion failed: (BrOffset <= 0) && "Wrong hardware loop", file llvm-project\llvm\lib\Target\Xtensa\XtensaInstrInfo.cpp, line 321
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: build\Debug\bin\llc.exe -filetype=obj reproducer.ll -O1
1.      Running pass 'Function Pass Manager' on module 'reproducer.ll'.
2.      Running pass 'Branch relaxation pass' on function 
```

aykevl, 2021-02-09:
Yes, this is the crash log I'm getting:

```
$ llc -filetype=obj reproducer.ll -O1
llc: /home/ayke/src/github.com/tinygo-org/tinygo/llvm-project/llvm/lib/Target/Xtensa/XtensaInstrInfo.cpp:321: virtual bool llvm::XtensaInstrInfo::isBranchOffsetInRange(unsigned int, int64_t) const: Assertion `(BrOffset <= 0) && "Wrong hardware loop"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.	Program arguments: /home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc -filetype=obj reproducer.ll -O1 
1.	Running pass 'Function Pass Manager' on module 'reproducer.ll'.
2.	Running pass 'Branch relaxation pass' on function '@crashing'
 #0 0x0000000002ea71b4 PrintStackTraceSignalHandler(void*) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x2ea71b4)
 #1 0x0000000002ea4d8e llvm::sys::RunSignalHandlers() (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x2ea4d8e)
 #2 0x0000000002ea74d5 SignalHandler(int) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x2ea74d5)
 #3 0x00007f14bca7d730 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x12730)
 #4 0x00007f14bc5af7bb raise (/lib/x86_64-linux-gnu/libc.so.6+0x377bb)
 #5 0x00007f14bc59a535 abort (/lib/x86_64-linux-gnu/libc.so.6+0x22535)
 #6 0x00007f14bc59a40f (/lib/x86_64-linux-gnu/libc.so.6+0x2240f)
 #7 0x00007f14bc5a8102 (/lib/x86_64-linux-gnu/libc.so.6+0x30102)
 #8 0x0000000001e1e5a6 llvm::XtensaInstrInfo::isBranchOffsetInRange(unsigned int, long) const (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x1e1e5a6)
 #9 0x0000000002185025 (anonymous namespace)::BranchRelaxation::isBlockInRange(llvm::MachineInstr const&, llvm::MachineBasicBlock const&) const (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x2185025)
#10 0x00000000021836d5 (anonymous namespace)::BranchRelaxation::runOnMachineFunction(llvm::MachineFunction&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x21836d5)
#11 0x00000000022e8b9d llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x22e8b9d)
#12 0x00000000026cfce4 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x26cfce4)
#13 0x00000000026d6418 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x26d6418)
#14 0x00000000026d0534 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x26d0534)
#15 0x000000000170a3e9 main (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x170a3e9)
#16 0x00007f14bc59c09b __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2409b)
#17 0x0000000001707e8a _start (/home/ayke/src/github.com/tinygo-org/tinygo/llvm-build/bin/llc+0x1707e8a)
```

closed 2021-03-05 by espressif-bot

andreisfr, 2021-03-05:
Hi @MabezDev , @aykevl ! Thank you very much for help! We fixed this issue in LLVM, please use latest version

commit 96776599 (2021-03-05): [Xtensa] Fix Hardware Loop optimization

Closes #43