espressif/llvm-project #45: Recent SPIRAM workarounds

lexxvir, 2021-02-10:
Hi,

As I know there are issues with external SPIRAM on ESP32 rev. 1.
Also the recent GCC (xtensa-esp32-elf-1.22.0-97-gc752ad5-5.2.0) has additional workarounds for these issues.

Does LLVM has these recent workarounds required for SPIRAM on ESP32 rev. 1?

Thanks

andreisfr, 2021-02-19:
Hi @lexxvir ! Please check last version, it includes PSRAM fix option like in gcc.

lexxvir, 2021-03-12:
Hi @andreisfr !

Thank you for the fix.
But it seems it has a flaw.

When I compiling a rust crate I get segmentation fault.
Not all crates affected by the segfault, but that are big enough.

I don't figured yet what is cause of the segfault but I've got the backtrace from the `rustc`:
```
Thread 2.7 "rustc" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe486e700 (LWP 7510)]
0x00007ffff031b460 in createXtensaPSRAMCacheFix::xtensaPSRAMCacheFixMemwReorg(llvm::MachineFunction&) ()
   from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
(gdb) bt
#0  0x00007ffff031b460 in createXtensaPSRAMCacheFix::xtensaPSRAMCacheFixMemwReorg(llvm::MachineFunction&) ()
   from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#1  0x00007ffff19a0728 in llvm::MachineFunctionPass::runOnFunction(llvm::Function&) () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#2  0x00007ffff2680dcf in llvm::FPPassManager::runOnFunction(llvm::Function&) () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#3  0x00007ffff26814c1 in llvm::FPPassManager::runOnModule(llvm::Module&) () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#4  0x00007ffff2680001 in llvm::legacy::PassManagerImpl::run(llvm::Module&) () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#5  0x00007ffff0306f30 in LLVMRustWriteOutputFile () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#6  0x00007ffff026fe7c in rustc_codegen_llvm::back::write::write_output_file () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#7  0x00007ffff0278990 in <rustc_codegen_llvm::LlvmCodegenBackend as rustc_codegen_ssa::traits::write::WriteBackendMethods>::codegen ()
   from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#8  0x00007ffff0257c0c in rustc_codegen_ssa::back::write::finish_intra_module_work () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#9  0x00007ffff0251a5a in rustc_codegen_ssa::back::write::execute_work_item () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#10 0x00007ffff01367bf in std::sys_common::backtrace::__rust_begin_short_backtrace () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#11 0x00007ffff01b76a5 in core::ops::function::FnOnce::call_once{{vtable-shim}} () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/librustc_driver-4564ebe630271a03.so
#12 0x00007fffef13f2f8 in std::sys::unix::thread::Thread::new::thread_start () from /home/aarbuzov/.rustup/toolchains/xtensa-1.48.0-spiram/lib/libstd-5ef3b15e433fd6bd.so
#13 0x00007fffee45a6db in start_thread (arg=0x7fffe486e700) at pthread_create.c:463
#14 0x00007fffeedbc71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

I provide following flags to LLVM: -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw
Without that flags compilation is fine.

Let me know if you need more info.

andreisfr, 2021-03-12:
Hi, @lexxvir !

Could I ask for help with this issues? Is it possible to localize problem in code in https://github.com/espressif/llvm-project/issues/43 or maybe you could describe steps to reproduce the problem?

lexxvir, 2021-03-15:
Hi @andreisfr !

I reduced the problem to following code:
```
; ModuleID = '<stdin>'
source_filename = "spin.3a1fbbbh-cgu.0"
target datalayout = "e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32"
target triple = "xtensa-none-unknown-elf"

%"core::sync::atomic::AtomicBool" = type { [0 x i8], i8, [0 x i8] }

; Function Attrs: minsize nounwind optsize
define void @str() unnamed_addr #0 {
start:
  %m = alloca %"core::sync::atomic::AtomicBool", align 4
  %0 = getelementptr inbounds %"core::sync::atomic::AtomicBool", %"core::sync::atomic::AtomicBool"* %m, i32 0, i32 0, i32 0
  call void @llvm.lifetime.start.p0i8(i64 1, i8* nonnull %0)
  store i8 0, i8* %0, align 4
  %1 = cmpxchg weak i8* %0, i8 0, i8 1 acquire monotonic
  call void @llvm.lifetime.end.p0i8(i64 1, i8* nonnull %0)
  ret void
}

; Function Attrs: argmemonly nounwind willreturn
declare void @llvm.lifetime.start.p0i8(i64 immarg, i8* nocapture) #1

; Function Attrs: argmemonly nounwind willreturn
declare void @llvm.lifetime.end.p0i8(i64 immarg, i8* nocapture) #1

attributes #0 = { minsize nounwind optsize "target-cpu"="esp32" }
attributes #1 = { argmemonly nounwind willreturn }
```

Save it as `spin.ll` and run the command:
`llc -filetype=obj ./spin.ll -o spin.o -mfix-esp32-psram-cache-issue`.

I get the segfault:
```
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.	Program arguments: /home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc -filetype=obj ./spin.ll -o spin.o -mfix-esp32-psram-cache-issue 
1.	Running pass 'Function Pass Manager' on module './spin.ll'.
2.	Running pass 'Xtensa fix PSRAM cache issue in the ESP32 chips' on function '@str'
 #0 0x000056309b6d0cba llvm::sys::PrintStackTrace(llvm::raw_ostream&) (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1730cba)
 #1 0x000056309b6cea64 llvm::sys::RunSignalHandlers() (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x172ea64)
 #2 0x000056309b6cebb3 SignalHandler(int) (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x172ebb3)
 #3 0x00007fd6b7f40980 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x12980)
 #4 0x000056309a7ec3a0 createXtensaPSRAMCacheFix::xtensaPSRAMCacheFixMemwReorg(llvm::MachineFunction&) (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x84c3a0)
 #5 0x000056309ac59818 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0xcb9818)
 #6 0x000056309b002d0f llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1062d0f)
 #7 0x000056309b003401 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1063401)
 #8 0x000056309b001f41 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x1061f41)
 #9 0x000056309a35714f main (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x3b714f)
#10 0x00007fd6b6dfebf7 __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:344:0
#11 0x000056309a3b541a _start (/home/aarbuzov/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x41541a)
zsh: segmentation fault (core dumped)  ~/git/rust/rust/build/x86_64-unknown-linux-gnu/llvm/bin/llc -filetype=obj  -o
```

Without `-mfix-esp32-psram-cache-issue` code compiles fine.

andreisfr, 2021-03-17:
Hi, @lexxvir ! Thank you very much! I'm able to reproduce the problem now, we will investigate it.

lexxvir, 2021-04-16:
Hi @andreisfr ! Are there any updates regarding this issue? Thank you.

andreisfr, 2021-04-19:
Hi @lexxvir !

We are preparing the commit which will correct the issue.

andreisfr, 2021-08-20:
Hi @lexxvir , the problem if fixed in new branch xtensa_release_12.0.1, please check it.

igrr, 2021-09-23:
@lexxvir That's about the same percentage increase as we see for the GCC based toolchain, so seems to be expected.

lexxvir, 2021-09-23:
Ok, thanks @igrr. So I close this for now.

closed 2021-09-23 by lexxvir