espressif/llvm-project #58: Inline assembly memory operands are broken

aykevl, 2022-04-15:
With this C file (`test.c`):

```c
int foo(int *ptr) {
        int n;
        __asm__ __volatile__("l32i %0, %1" : "=r"(n) : "m"(*ptr));
        return n;
}
```

I get the following error:

```
$ llvm-build/bin/clang -c -o tmp/test.o -Os tmp/test.c --target=xtensa
tmp/test.c:3:23: error: unexpected token
        __asm__ __volatile__("l32i %0, %1" : "=r"(n) : "m"(*ptr));
                             ^
<inline asm>:1:12: note: instantiated into assembly here
        l32i a2, 0(a2)
                  ^
1 error generated.
```

It looks like the inline assembly is incorrect. It it should have emitted `l32i a2, a2, 0` instead of `l32i a2, 0(a2)`.

andreisfr, 2022-05-19:
@aykevl , thank you for your investigation of this issue, now it is fixed, please use xtensa_release_14.0.0 branch with latest changes

aykevl, 2022-05-21:
I can confirm that ee16f0217f615ea7fea3cd018906182bf7bb6638 fixes this issue. Thank you!
Feel free to close this issue.

closed 2022-08-11 by igrr