espressif/llvm-project #60: `rust-bindgen` can't use `libclang.dll`

bugadani, 2022-04-29:
I'm trying to use the xtensa-enabled llvm 14 fork together with `rust-bindgen`. Unfortunately, it fails to load the `libclang.dll` with the following message: `Unable to find libclang: "the `libclang` shared library at C:\\Users\\bugad\\.llvm\\xtensa-esp32-elf-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"`

This seems to work well with the upstream llvm.

bugadani, 2022-04-29:
After a bit of googling, it looks like an issue with how I'm setting things up.

closed 2022-04-29 by bugadani

bugadani, 2023-12-02:
@lhns I can't say that I remember well but at least part of the solution was to point LIBCLANG_PATH to `<your home>\toolchains\esp\xtensa-esp32-elf-clang\esp-16.0.0-20230516\esp-clang\bin\libclang.dll`

lhns, 2023-12-02:
I now also got it working by modifying the PATH variable and putting the toolchain bin paths in the first place.