espressif/llvm-project #61: Release aarch64 artifacts for LLVM14

kevin-june, 2022-08-04:
The [LLVM13 release](https://github.com/espressif/llvm-project/releases/tag/esp-13.0.0-20211203) contains artifacts for:
* aarch64-apple-darwin
* aarch64-unknown-linux-gnu

We would like to use similar artifacts in [LLVM 14](https://github.com/espressif/llvm-project/releases/tag/esp-14.0.0-20220415), but these don't seem to be present.

Is it possible to add these to the release?

igrr, 2022-08-05:
Hi @kevin-june,
The artifacts for aarch64 have been built for LLVM13 release manually by the esp-rs team. For LLVM14 the archives have also been built manually and uploaded to https://github.com/esp-rs/rust-build/releases/tag/llvm-project-14.0-minified instead, to avoid confusion.

We have recently merged support for building aarch64-unknown-linux-gnu releases in CI, and CI builds for aarch64-apple-darwin are in development. So the next llvm-project release should have archives for aarch64 built. For now, please try to use the archives provided in the rust-build repo.

gerekon, 2022-08-05:
@kevin-june @igrr FYI. The artifacts in esp-rs are stripped version of the release which include libraries and headers only used by Rust.

kevin-june, 2022-08-05:
Thanks for the info!

Will the future release include standard library headers? We had been using these to satisfy Bindgen.

gerekon, 2022-08-08:
> Thanks for the info!
> 
> Will the future release include standard library headers? We had been using these to satisfy Bindgen.

Are you talking about special stripped (`minified`) releases for Rust?
What standard library do you mean, stdc++?

gerekon, 2022-09-24:
@kevin-june We made a pre-release with ARM64 artidacts

closed 2022-09-24 by gerekon

gerekon, 2022-09-24:
Closed by e95d783d2a30cfbdf4694a7dd3903166861f638c and 789d4979c11840cd9d2626c89234ac51409c80c6

kevin-june, 2022-09-26:
Thanks!

> What standard library do you mean, stdc++?

We primarily use C headers.