espressif/llvm-project #66: Calling convention for arguments on the stack on Xtensa is incorrect for arguments narrower than 32 bits

kyrias, 2023-04-17:
According to the ISA document function arguments passed on the stack are all essentially stored in 32-bit slots, which matches how GCC for Xtensa behaves. Meanwhile LLVM for Xtensa passes smaller arguments packed together.  This means that passing more than 6 arguments to a function between code compiled by GCC and LLVM is currently broken.

E.g.:

```
GCC:
| 8-bit  | padding                | 16-bit         | padding        | 32-bit                         |

LLVM:
| 8-bit  | 16-bit         | 32-bit                         |
```

I see in `CC_Xtensa_Custom` that `i8` and `i16` are promoted to `i32` for non-byval arguments:

https://github.com/espressif/llvm-project/blob/1e28f73ab9cb70f289f3eb25bb7875ba8694da0c/llvm/lib/Target/Xtensa/XtensaISelLowering.cpp#L744-L753

But the same promotion s not being done for byval ones:

https://github.com/espressif/llvm-project/blob/1e28f73ab9cb70f289f3eb25bb7875ba8694da0c/llvm/lib/Target/Xtensa/XtensaISelLowering.cpp#L733-L742

---

I was considering trying to submit a patch for this, but I'm not sure if you're currently accepting patches for the Xtensa part, or if so _how_ you're accepting them.

kyrias, 2023-04-24:
@andreisfr Sorry for the direct ping but it seems like you committed the relevant code.  Do you have any idea of what kind of timeline we're looking at to get this fixed?  Would be good to have this ABI incompatibility resolved.

igrr, 2023-04-24:
This issue has also been reported internally last week by @MabezDev and the fix is now in progress (internal tracker is LLVM-244). Sorry for not leaving a comment here in time. I think we will have the fix merged this or worst case next week.

kyrias, 2023-05-08:
@igrr Ping?

andreisfr, 2023-05-09:
Hi @kyrias , we preparing patch now with fixes, I hope in a few days we will update current version.

andreisfr, 2023-05-10:
@kyrias , we applied fixes, PTAL.

kyrias, 2023-05-10:
@andreisfr The generated code from the `xtensa_release_15.x` branch looks correct now.  :+1:

gerekon, 2023-05-16:
Should be fixed in df983770e12eee54ee0228630be82405ea21934c and included into [esp-16.0.0-20230515](https://github.com/espressif/llvm-project/releases/tag/esp-16.0.0-20230515) release.

@kyrias Can we close this issue?

closed 2023-05-17 by kyrias