espressif/llvm-project #72: tail call

yamt, 2023-06-15:
there is a todo comment about tail call optimization:
https://github.com/espressif/llvm-project/blob/6ef268936af6df777f165827ecc97a717b95f9fc/llvm/lib/Target/Xtensa/XtensaISelLowering.cpp#L1039-L1040

do you have any plans or ideas about implementation?
i understand that it isn't straightforward with the windowed abi.

yamt, 2023-06-16:
the current code just ignores the `musttail` attribute.
it should bail out at least.
```
void
f(void)
{
        void g(void);
        __attribute__((musttail)) return g();
}
```

yamt, 2023-06-16:
> the current code just ignores the `musttail` attribute. it should bail out at least.
> 
> ```
> void
> f(void)
> {
>         void g(void);
>         __attribute__((musttail)) return g();
> }
> ```

https://github.com/espressif/llvm-project/pull/73

andreisfr, 2023-06-20:
Hi @yamt ! Thank you for comment, we have a plans about "tail call" implementation.

yamt, 2023-06-25:
> Hi @yamt ! Thank you for comment, we have a plans about "tail call" implementation.

are you implying a general implementation is possible? how? just curious.

when caller and callee happen to have the same frame size, maybe you can make it jump into callee+3.
otherwise, i'm not sure how it can be done.

ProfFan, 2024-05-06:
Hello @andreisfr do you know what is the status for tail call?