espressif/llvm-project #74: `{,u}int_fast8_t` incompatibility between GCC and LLVM

kyrias, 2023-06-16:
There appears to be an ABI incompatibility between GCC and LLVM on Xtensa when using `int_fast8_t` and `uint_fast8_t`.

I installed both the GCC and LLVM toolchains using the following command:

```
espup install --toolchain-version 1.69.0.1 --targets esp32s3 --extended-llvm
```

Using the following resulting binaries:
- GCC: `~/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc`
- Clang: `~/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/bin/clang`
  The Clang target was set to `xtensa-esp32s3-espidf`.

Looking at the assembly emitted from the following C program I see that GCC considers it 4 bytes and Clang 1.

```c
#include <stdint.h>

int main(void) {
    return (sizeof (uint_fast8_t));
}
```

<details>
<summary>Assembly from GCC</summary>

```asm
	.file	"foo.c"
	.text
	.section	.text.startup,"ax",@progbits
	.align	4
	.global	main
	.type	main, @function
main:
	entry	sp, 32
	movi.n	a2, 4
	retw.n
	.size	main, .-main
	.ident	"GCC: (crosstool-NG esp-12.2.0_20230208) 12.2.0"
```
</details>

<details>
<summary>Assembly from Clang</summary>

```asm
	.text
	.file	"foo.c"
	.global	main                            # -- Begin function main
	.p2align	2
	.type	main,@function
main:                                   # @main
# %bb.0:
	entry	a1, 32
	movi.n	a2, 1
	retw.n
.Lfunc_end0:
	.size	main, .Lfunc_end0-main
                                        # -- End function
	.ident	"clang version 16.0.1 (https://github.com/espressif/llvm-project.git fe4f10a809705c2ddbed62145616bb9c85c499c0)"
	.section	".note.GNU-stack","",@progbits
	.addrsig
```
</details>

I have not checked whether

kyrias, 2023-06-16:
Also, are you aware that every time someone opens an issue on this issue tracker they get a CI job failure from GitHub due to the mailgun job requiring a key parameter?  It's a bit annoying, though even if that was fixed I guess I'd get a success email instead...

andreisfr, 2023-06-20:
Hi @kyrias ! Thank you for posting the problem, we investigate it.

andreisfr, 2023-11-29:
Hi @kyrias ! Could you try to use latest changes in branch xtensa_release_16.0.4?
We implemented "-mfast-int-min32" option, which sets minimum size to _FAST_ int types in clang to 4 bytes.

closed 2023-12-12 by andreisfr