espressif/llvm-project #75: Rust builds failing with LLVM ERROR: Unexpected operand value!

dacut, 2023-06-23:
Right now, I don't have a small, reproducible build to show here. I'm opening this issue as a placeholder for others to search in case they run into this issue and can view my progress in tracking down where things are going awry.

Currently, to reproduce, here are the steps (assuming you have the Rust toolchain installed):
* Check out ESP-IDF v5.1:  
    ```sh
    mkdir -p ~/esp
    git clone --depth 1 --branch release/v5.1 https://github.com/espressif/esp-idf.git ~/esp/esp-idf
    ```
* Set up the Rust build environment to target ESP32:
    ```sh
    cargo install espup
    espup install
    cargo install ldproxy
    source ~/export-esp.sh
    export IDF_PATH=$HOME/esp/esp-idf
    ```
* Check out the Rust ESP-IDF-HAL repository:
    ```sh
    git clone https://github.com/esp-rs/esp-idf-hal ~/esp-idf-hal
    ```
* Attempt to build:
    ```sh
    cargo build --target xtensa-esp32-espidf -Zbuild-std=std,panic_abort
    ```

The build will fail with:
```
   Compiling esp-idf-hal v0.41.1 (/home/dacut/projects/esp-idf-hal)
LLVM ERROR: Unexpected operand value!
```

I have further instrumented logging to get:
```
LLVM ERROR: Unexpected operand value! getMemRegEncoding2: opcode=418, Res=0x21
```
Which means it's failing [here](https://github.com/espressif/llvm-project/blob/xtensa_release_16.0.4/llvm/lib/Target/Xtensa/MCTargetDesc/XtensaMCCodeEmitter.cpp#L315), trying to emit an L32I_N instruction against a non-aligned value. The error is obviously elsewhere (where the instruction is being encoded in the first place).

Right now, I'm tracking down where this was introduced using `git bisect` and will continue to report progress here.

dacut, 2023-06-24:
To work around this, for now the solution is to downgrade your ESP toolchain (thanks @mabezdev!):
```sh
espup install --toolchain-version 1.69.0
```

andreisfr, 2023-06-28:
Hi @dacut ! Thank you for report. We will investigate the problem. BTW we found similar error and in a week we will upload fixes to main branch, it is high probability that reported problem will be also fixed.

dacut, 2023-06-29:
Hi @andreisfr! I can confirm that this is being worked on, per conversations on the Matrix server, and expected to land in the 1.71 toolchain. I opened this so others could find it more easily (since we can't easily tag Matrix conversations in Git 😄 ).

andreisfr, 2023-07-17:
Hi @dacut ! Could you verify the latest version of the main branch. I hope this commit https://github.com/espressif/llvm-project/commit/2aba83d226248772cf590889ae2a6a189d0d8c4d should correct the problem.

dacut, 2023-07-21:
Hi @andreisfr! This is now working indeed. Thanks!

closed 2023-07-21 by dacut