espressif/llvm-project #79: clang/llvm15: spurious dereference when loading float from literal pool

q3k, 2023-07-28:
Take the following example code:

```
$ cat repro.c
float ctx_maxf (float a, float b) {
    return ((float)(a > b)) * a + ((float)(a <= b)) * b;
}
```

This generates the following assembly:

```
$ clang repro.c -o repro.clang -nostartfiles -O2  --target=xtensa-esp-elf -mcpu=esp32s3
$ xtensa-esp32s3-elf-objdump -d  repro.clang

repro.clang:     file format elf32-xtensa-le

Disassembly of section .text:

00400074 <ctx_maxf-0x8>:
  400074:	000000        	ill
  400077:	000000        	ill
  40007a:	80          	.byte 0x80
  40007b:	3f          	.byte 0x3f

0040007c <ctx_maxf>:
  40007c:	004136        	entry	a1, 32
  40007f:	fa9350        	wfr	f9, a3
  400082:	fa8250        	wfr	f8, a2
  400085:	090c      	movi.n	a9, 0
  400087:	180c      	movi.n	a8, 1
  400089:	6b0890        	ole.s	b0, f8, f9
  40008c:	08bd      	mov.n	a11, a8
  40008e:	011076        	bt	b0, 400093 <ctx_maxf+0x17>
  400091:	09bd      	mov.n	a11, a9
  400093:	fff8a1        	l32r	a10, 400074 <ctx_maxf-0x8> (0 <ctx_maxf-0x40007c>)
  400096:	a0bba0        	addx4	a11, a11, a10
  400099:	000ba3        	lsi	f10, a11, 0
  40009c:	2aa9a0        	mul.s	f10, f9, f10
  40009f:	4b0980        	olt.s	b0, f9, f8
  4000a2:	011076        	bt	b0, 4000a7 <ctx_maxf+0x2b>
  4000a5:	098d      	mov.n	a8, a9
  4000a7:	a088a0        	addx4	a8, a8, a10
  4000aa:	000893        	lsi	f9, a8, 0
  4000ad:	4aa980        	madd.s	f10, f9, f8
  4000b0:	fa2a40        	rfr	a2, f10
  4000b3:	f01d      	retw.n
```

Or, equivalent decompiled code (note the `*(float *)`):

```
float ctx_maxf(float param_1,float param_2)

{
  return param_2 * *(float *)((uint)(param_1 <= param_2) * 4) +
         *(float *)((uint)(param_2 < param_1) * 4) * param_1;
}
```

This code crashes at ` 400099:	000ba3        	lsi	f10, a11, 0` with a11 equal 0 or 4 (instead of 400074 or 400078).

```
$ clang --version
clang version 15.0.0 (https://github.com/espressif/llvm-project.git 23786128aea96880a5752f6bd1c22aab27622ccf)
Target: x86_64-unknown-linux-gnu
Thread model: posix
```

andreisfr, 2023-08-02:
Hello @q3k ! Thank you very much for your investigation. We work on this problem now

andreisfr, 2023-08-04:
@q3k, could you try to use latest version Xtensa llvm (16.0.4) to compile this example? It seems the problem with litera sections is fixed in it.

closed 2023-09-01 by gerekon