espressif/llvm-project #81: Disassembly files showing no instructions being executed.

rohitpreddy07, 2023-08-07:
I'm using the default release repo for my LLVM compiler build to generate code for xtensa-specific architecture.  However, I'm facing issues running the executable using the xt-clang command xt-run. Observing the .dis file for my code it seems that there are no instructions in the main, hence it is not executing. Does anyone know a quick fix?`//main.c
#include "f.h"

int main(){
    
    int c= fun(10,12);
    
    return 0;
}`

`//f.h
int fun(int a,int b){
  int c= a+b;
  return c;
}
`

andreisfr, 2023-08-07:
Hi @rohitpreddy07 ! I'm sorry, did you use xtensa_release_16.0.4 branch from this repository to build clang, or some other version? As I'm understand, xt-clang is version provided by Cadence?

rohitpreddy07, 2023-08-08:
@andreisfr I'm generating the .obj files using clang built from xtensa_release_16.0.4 branch and linking and running the executable using xt-clang provided by Cadence. Just doing some experimenting of my own.

rohitpreddy07, 2023-08-08:
I compiled the same code using xtensa branch and xt-clang and got these outputs in the .dis files.
This is the main section for the .dis file using xt-clang:
`80000578 <main>:
80000578:	006136        	entry	a1, 48
8000057b:	012d      	mov.n	a2, a1
8000057d:	1129      	s32i.n	a2, a1, 4
8000057f:	aa0c      	movi.n	a10, 10
80000581:	cb0c      	movi.n	a11, 12
80000583:	0000a5        	call8	8000058c <_Z3funii>
80000586:	01a9      	s32i.n	a10, a1, 0
80000588:	020c      	movi.n	a2, 0
8000058a:	f01d      	retw.n`

This is the same using the xtensa branch:
`80000240 <main>:
80000240:	7d006136 29020c01 0caa0c17 ff9881cb     6a.}...)........
80000250:	a90008e0 00f01d07                       ........`

Interestingly, I can see the entry and retw.n opcodes in the branch output however they seem to be shifted a few bytes. Could this be a reason why the tools don't seem to be executing properly?

andreisfr, 2023-08-08:
May I ask you to attach to attach object file produced by xt-clang?

rohitpreddy07, 2023-08-09:
Sure! 
[test.o.zip](https://github.com/espressif/llvm-project/files/12298530/test.o.zip)

closed 2023-09-11 by rohitpreddy07