espressif/llvm-project #90: indirect calls are always used

yamt, 2024-02-06:
currently, -mlongcalls is always effectively on.
i had to patch llvm for my use case. cf. https://github.com/yamt/llvm-project/commit/3f04244a4fd94952a3f208b7d1ee3afef2da1e3a
it would be nicer to have a way to use direct calls.
ideally, the backend can automatically choose direct calls when the call distance is not too long, i guess.

yamt, 2024-02-16:
for my use case, it's ok to manually adding attributes at IR level. eg. xtensa-short-call.
mips has something along the line: https://clang.llvm.org/docs/AttributeReference.html#short-call-near

gerekon, 2024-02-16:
@yamt Thanks for pointing this out. We are working on direct calls support using `callN` instruction.

cc @sstefan1

sstefan1, 2024-02-19:
> for my use case, it's ok to manually adding attributes at IR level. eg. xtensa-short-call. mips has something along the line: https://clang.llvm.org/docs/AttributeReference.html#short-call-near

We're working on implementing the support for direct calls, the same way gcc does it. That is a bit more complex. In the mean time we can implement `short-call` attributes for xtensa.

dongsheng28849455, 2024-04-17:
hi, @sstefan1, can I ask the progress for this feature?

dongsheng28849455, 2024-05-13:
> > hi, @sstefan1, can I ask the progress for this feature?
> 
> There hasn't been any real progress since that coment. I'll write here when I have some updates.

is that fixed by https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240408?

yamt, 2024-05-13:
> > > hi, @sstefan1, can I ask the progress for this feature?
> > 
> > 
> > There hasn't been any real progress since that coment. I'll write here when I have some updates.
> 
> is that fixed by https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240408?

the short-call attribute is already in the releases and it seems working as far as i tested.
the "ideal" solution (ie. somehow automatically use short/long calls) is not there yet.
i guess he was talking about the latter.

gerekon, 2024-05-14:
> the short-call attribute is already in the releases and it seems working as far as i tested. the "ideal" solution (ie. somehow automatically use short/long calls) is not there yet. i guess he was [...]

Yes, exactly. Automatic use of short/long calls is in progress. 
Just for information: We do not expect it to be ready in the 1st half of this year.

andyross, 2024-06-10:
Just bumped into this myself.  Is there no support for at least changing the default e.g. ("-mno-longcalls")?  The range of a CALLx instruction is +/- 512k, which is quite a bit larger than the .text segment of many/most firmware images in the wild.  Right now the conservative default is a pretty severe performance regression, and it's needless for a lot of applications.

andreisfr, 2024-06-12:
@andyross , thank you for the question. Currently we work on implementing CALLX support in the LLD.