espressif/llvm-project #97: run-clang-tidy.py not found in your PATH

greentd, 2024-06-17:
esp-clang Has been successfully installed, 

D:\esptest\11\hello_world>clang --version
clang version 17.0.1 (https://github.com/espressif/llvm-project.git esp-17.0.1_20240419)
Target: riscv32-esp-unknown-elf
Thread model: posix
InstalledDir: D:/esp32/espressif/tools/esp-clang/esp-17.0.1_20240419/esp-clang/bin

run    D:\esptest\hello_world>idf.py clang-check

but not generated warning.txt

Error message

****** Filter files and dirs ******
Files to be analysed:
+ > D:\esptest\hello_world\main\hello_world_main.c
***********************************
run-clang-tidy.py not found in your PATH

How to solve?

gerekon, 2024-06-21:
Starting from some version (maybe LLVM 16) `run-clang-tidy.py` is installed as `run-clang-tidy`. 
It looks like the problem of `idf.py` script.
As a workaround you can rename `run-clang-tidy` to `run-clang-tidy.py` in your Clang toolchain installation.

dobairoland, 2024-06-21:
@radimkarnis PTAL

radimkarnis, 2024-07-24:
> As a workaround you can rename run-clang-tidy to run-clang-tidy.py in your Clang toolchain installation.

[clang-tidy-runner](https://github.com/espressif/clang-tidy-runner) looks both for `run-clang-tidy` and `run-clang-tidy.py`, this shouldn't be an issue.

@greentd I have tried to reproduce the issue - and succeeded only if my installation was incomplete. Going through the [install process](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-clang-tidy.html) again solves the issue. Can you please try that?

Should be as easy as:
1) Running the ESP-IDF install script
2) Running `idf_tools.py install esp-clang`
3) Running the ESP-IDF export script

gerekon, 2024-08-23:
Closing as we have not received feedback from the user.

closed 2024-08-23 by gerekon