espressif/llvm-project #98: broken dwarf

yamt, 2024-06-21:
the following is with `esp-17.0.1_20240419` built locally.
at least, DW_AT_producer, DW_AT_name, DW_AT_decl_file, DW_AT_comp_dir seem broken.

```shell
spacetanuki% cat a.c
int f(int x) { return x; }
spacetanuki% /Volumes/PortableSSD/llvm-xtensa/llvm-project/build/bin/clang -c -Os -g a.c
spacetanuki% llvm-dwarfdump a.o                                                          
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
warning: failed to compute relocation: R_XTENSA_32, Invalid data was encountered while parsing the file
a.o: Error in creating MCRegInfo  
a.o:    file format elf32-xtensa                                              
                                       
.debug_info contents:                                                         
0x00000000: Compile Unit: length = 0x0000003e, format = DWARF32, version = 0x0005, unit_type = DW_UT_compile, abbr_offset = 0x0000, addr_size = 0x04 (next un
it at 0x00000042)                     
error: invalid reference to or invalid content in .debug_str_offsets[.dwo]: insufficient space for 32 bit header prefix
                                       
0x0000000c: DW_TAG_compile_unit
              DW_AT_producer    ()
              DW_AT_language    (DW_LANG_C11)
              DW_AT_name        ()
              DW_AT_str_offsets_base    (0x00000000)
              DW_AT_stmt_list   (0x00000000)
              DW_AT_comp_dir    ()
              DW_AT_low_pc      (0x00000008)
              DW_AT_high_pc     (0x0000000d)
              DW_AT_addr_base   (0x00000000)

0x00000023:   DW_TAG_subprogram
                DW_AT_low_pc    (0x00000008)
                DW_AT_high_pc   (0x0000000d)
                DW_AT_frame_base        (DW_OP_reg1)
                DW_AT_call_all_calls    (true)
                DW_AT_name      ()
                DW_AT_decl_file ("/tmp/t7")
                DW_AT_decl_line (1)
                DW_AT_prototyped        (true)
                DW_AT_type      (0x0000003d "base ")
                DW_AT_external  (true)

0x00000032:     DW_TAG_formal_parameter
                  DW_AT_location        (DW_OP_reg2)
                  DW_AT_name    ()
                  DW_AT_decl_file       ("/tmp/t7")
                  DW_AT_decl_line       (1)
                  DW_AT_type    (0x0000003d "base ")

0x0000003c:     NULL

0x0000003d:   DW_TAG_base_type
                DW_AT_name      ()
                DW_AT_encoding  (DW_ATE_signed)
                DW_AT_byte_size (0x04)

0x00000041:   NULL
spacetanuki% 
```

gerekon, 2024-06-21:
Hi @yamt 

How do you compile clang?
I do not see his problem with our latest made st release for this tag. 
```bash
lp_uart_print (feat/ulp_debug*) » clang -c -Os -g /tmp/a.c 
lp_uart_print (feat/ulp_debug*) » llvm-dwarfdump a.o
a.o:	file format elf32-littleriscv

.debug_info contents:
0x00000000: Compile Unit: length = 0x0000004d, format = DWARF32, version = 0x0004, abbr_offset = 0x0000, addr_size = 0x04 (next unit at 0x00000051)

0x0000000b: DW_TAG_compile_unit
              DW_AT_producer	("Espressif clang version 17.0.1 (https://github.com/espressif/llvm-project.git esp-17.0.1_20240419)")
              DW_AT_language	(DW_LANG_C11)
              DW_AT_name	("/tmp/a.c")
              DW_AT_stmt_list	(0x00000000)
              DW_AT_comp_dir	("/home/alexey/projects/esp/esp-idf/examples/system/ulp/lp_core/lp_uart/lp_uart_print")
              DW_AT_low_pc	(0x00000000)
              DW_AT_high_pc	(0x00000002)

0x00000026:   DW_TAG_subprogram
                DW_AT_low_pc	(0x00000000)
                DW_AT_high_pc	(0x00000002)
                DW_AT_frame_base	(DW_OP_reg2 X2)
                DW_AT_GNU_all_call_sites	(true)
                DW_AT_name	("f")
                DW_AT_decl_file	("/tmp/a.c")
                DW_AT_decl_line	(1)
                DW_AT_prototyped	(true)
                DW_AT_type	(0x00000049 "int")
                DW_AT_external	(true)

0x0000003b:     DW_TAG_formal_parameter
                  DW_AT_location	(DW_OP_reg10 X10)
                  DW_AT_name	("x")
                  DW_AT_decl_file	("/tmp/a.c")
                  DW_AT_decl_line	(1)
                  DW_AT_type	(0x00000049 "int")

0x00000048:     NULL

0x00000049:   DW_TAG_base_type
                DW_AT_name	("int")
                DW_AT_encoding	(DW_ATE_signed)
                DW_AT_byte_size	(0x04)

0x00000050:   NULL
```

yamt, 2024-06-21:
> Hi @yamt
> 
> How do you compile clang? I do not see this problem with our latest release made for this tag.

my configuration is:
```
cmake \
-G Ninja \
-S llvm -B ${BUILD_DIR} \
-DCMAKE_BUILD_TYPE=Debug  \
-DLLVM_ENABLE_PROJECTS="lld;clang;lldb" \
-DLLVM_TARGETS_TO_BUILD="WebAssembly;X86" \
-DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=Xtensa \
-DLLVM_DEFAULT_TARGET_TRIPLE="xtensa-esp32-elf" \
-DLLVM_INCLUDE_BENCHMARKS=OFF \
-DLLVM_INCLUDE_EXAMPLES=OFF \
-DLLVM_INCLUDE_TESTS=ON \
-DLLVM_INCLUDE_TOOLS=ON \
-DLLVM_ENABLE_ASSERTIONS=ON \
-DLLVM_ENABLE_TERMINFO=OFF \
-DLLVM_ENABLE_ZLIB=OFF \
-DLLVM_ENABLE_ZSTD=OFF \
-DLLVM_ENABLE_LIBXML2=OFF \
-DLLVM_ENABLE_LIBEDIT=OFF \
-DLLVM_HAVE_LIBXAR=OFF \
-DLLVM_INSTALL_BINUTILS_SYMLINKS=ON \
-DLLDB_INCLUDE_TESTS=OFF \
-DCMAKE_INSTALL_PREFIX=${HOME}/llvm
```

my colleague independently had a similar symptom (broken dwarf) with https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/wamr-compiler. i don't know how she built it.

yamt, 2024-06-21:
i was able to reproduce the symptom with the release assets.
https://github.com/espressif/llvm-project/releases/download/esp-17.0.1_20240419/clang-esp-17.0.1_20240419-x86_64-apple-darwin.tar.xz

```shell
clang --target=xtensa-esp-unknown-elf -c -Os -g a.c
```

yamt, 2024-06-21:
well, actully it turned out to be a problem in llvm-dwarfdump.
my fix: https://github.com/llvm/llvm-project/pull/96311

gerekon, 2024-06-26:
@yamt Thanks you very much for the findings. We will pick it up. So let's keep this ticket open until that moment .