espressif/llvm-project #100: How do you build a release?

MikeColeGuru, 2024-06-28:
Hi,

I am trying to compile a newer version of the esp-clang package for the platforms.

For example for version esp-17.0.1_20240419, the binary packages are here:
https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240419

I have compiled llvm/clang from the repo successfully.
However, I might be configuring llvm build wrong as the output does not include all the files included in the release.

For example these files and several others:

```
xtensa-esp32-elf-clang-as
xtensa-esp32-elf-clang-ld
xtensa-esp32-elf-clang-objcopy
xtensa-esp32-elf-clang-objdump
xtensa-esp32s2-elf-clang-as
xtensa-esp32s2-elf-clang-ld
xtensa-esp32s2-elf-clang-objcopy
xtensa-esp32s2-elf-clang-objdump
xtensa-esp32s3-elf-clang-as
xtensa-esp32s3-elf-clang-ld
xtensa-esp32s3-elf-clang-objcopy
xtensa-esp32s3-elf-clang-objdump
```

Is there a script somewhere (or instructions) that builds all of the releases for all of the platforms?

Thanks!

gerekon, 2024-06-28:
Hi!

We use Cmake scripts from this repo https://github.com/espressif/esp-llvm-embedded-toolchain. Unfortunately README does not contain Espressif specific instructions on how to build the toolchain. It is in my TODO list.

You can have a look at GitLab jobs used to build the toolchain 
- Core components: https://github.com/espressif/llvm-project/blob/xtensa_release_18.1.2/.gitlab-ci.yml#L217
- Target libraries: https://github.com/espressif/llvm-project/blob/xtensa_release_18.1.2/.gitlab-ci.yml#L223
- Pack job: https://github.com/espressif/llvm-project/blob/xtensa_release_18.1.2/.gitlab-ci.yml#L305. It dimply merges previous two steps into one archive.

Let me check it. I will try provide you with list of exact commands

gerekon, 2024-06-28:
You can try the following commands
```bash
git clone -b esp-17.0.1_20240419 https://github.com/espressif/esp-llvm-embedded-toolchain.git

GL_SERVER=github.com \
cmake $PWD/esp-llvm-embedded-toolchain -GNinja \
        -DLLVMPROJECT_REPO_URL="https://$GL_SERVER/espressif/llvm-project.git" \
        -DNEWLIB_REPO_URL="https://$GL_SERVER/espressif/newlib-esp32.git" \
        -DBINUTILS_REPO_URL="https://$GL_SERVER/espressif/binutils-gdb.git" \
        -DXTENSA_OVERLAYS_REPO_URL="https://$GL_SERVER/espressif/xtensa-overlays.git" \
        -DFETCHCONTENT_QUIET=OFF \
        -DESP_GNU_TOOLCHAIN_VER="13.2.0_20240305" \
        -DCPACK_ARCHIVE_THREADS=0 \
        -DUSE_LIBC=newlib \
        -DUSE_LIBCXX="libstdcxx" \
        -DUSE_RTLIB="compiler-rt;libgcc" \
        -DUSE_BINUTILS=ON \
        -DESP_TOOLCHAIN=ON \
        -DHOST_TRIPLE=x86_64-linux-gnu \
        -DLLVM_TOOLCHAIN_ENABLED_TARGETS="RISCV;Xtensa" \
        -B /tmp/build_esp_clang \
        --install-prefix=/tmp/esp-clang_Release

# install toolchain to /tmp/esp-clang_Release 
ninja -C /tmp/build_esp_clang install-llvm-toolchain

# or package toolchain. Archive should be in /tmp/build_esp_clang
ninja -C /tmp/build_esp_clang package-llvm-toolchain
```

gerekon, 2024-08-21:
Instructions are here https://github.com/espressif/esp-llvm-embedded-toolchain/blob/master/docs/building-from-source.md#espressif-toolchain-specifics.

closed 2024-08-21 by gerekon