espressif/llvm-project #103: rustc LLVM-18 float extension crash

MabezDev, 2024-10-21:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Targets with float support to compile

### Actual behavior (suspected bug)

It seems float extension i.e compiler support for going from `f16` -> `f32` is broken and produces the following LLVM error below.

### Error logs or terminal output

```shell
rustc-LLVM ERROR: Cannot select: 0x7ff58d078ee0: f32 = fp16_to_fp 0x7ff58c609d10
  0x7ff58c609d10: i32,ch = CopyFromReg 0x7ff58caa0fe0, Register:i32 %5
    0x7ff58c07dc10: i32 = Register %5
In function: _ZN17compiler_builtins5float6extend13__extendhfsf217h7e81549493d256faE
error: could not compile `compiler_builtins` (lib)
```
```

### Steps to reproduce the behavior

1) Download and unzip [ll.zip](https://github.com/user-attachments/files/17461741/ll.zip)
2) Using LLC from the latest release of llvm 18 run `./llc compiler_builtins-41e65ec39b07922e.compiler_builtins.6ff7e3b67cc0ddd4-cgu.0.rcgu.ll`
3) Observe an error like

```
LLVM ERROR: Cannot select: 0x565015758970: f32 = fp16_to_fp 0x56501563bec0
  0x56501563bec0: i32,ch = CopyFromReg 0x56501462f2c0, Register:i32 %5
    0x56501575ddb0: i32 = Register %5
In function: _ZN17compiler_builtins5float6extend13__extendhfsf217h7e81549493d256faE
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ./llc /home/mabez/development/rust/embedded/hal/esp-hal/examples/target/xtensa-esp32-none-elf/release/deps/compiler_builtins-41e65ec39b07922e.compiler_builtins.6ff7e3b67cc0ddd4-cgu.0.rcgu.ll
1.      Running pass 'Function Pass Manager' on module '/home/mabez/development/rust/embedded/hal/esp-hal/examples/target/xtensa-esp32-none-elf/release/deps/compiler_builtins-41e65ec39b07922e.compiler_builtins.6ff7e3b67cc0ddd4-cgu.0.rcgu.ll'.
2.      Running pass 'Xtensa DAG->DAG Pattern Instruction Selection' on function '@_ZN17compiler_builtins5float6extend13__extendhfsf217h7e81549493d256faE'
 #0 0x000056500e88d330 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (./llc+0x2d6a330)
 #1 0x000056500e88a77b SignalHandler(int) Signals.cpp:0:0
 #2 0x00007ffb18841960 (/lib64/libc.so.6+0x3c960)
 #3 0x00007ffb1889326c (/lib64/libc.so.6+0x8e26c)
 #4 0x00007ffb188418b6 raise (/lib64/libc.so.6+0x3c8b6)
 #5 0x00007ffb188298b7 abort (/lib64/libc.so.6+0x248b7)
 #6 0x000056500bde0e61 llvm::install_bad_alloc_error_handler(void (*)(void*, char const*, bool), void*) (.cold) ErrorHandling.cpp:0:0
 #7 0x000056500e63b398 llvm::SelectionDAGISel::CannotYetSelect(llvm::SDNode*) (./llc+0x2b18398)
 #8 0x000056500e63fc1b llvm::SelectionDAGISel::SelectCodeCommon(llvm::SDNode*, unsigned char const*, unsigned int) (./llc+0x2b1cc1b)
 #9 0x000056500d141dcd (anonymous namespace)::XtensaDAGToDAGISel::Select(llvm::SDNode*) XtensaISelDAGToDAG.cpp:0:0
#10 0x000056500e63715c llvm::SelectionDAGISel::DoInstructionSelection() (./llc+0x2b1415c)
#11 0x000056500e642fda llvm::SelectionDAGISel::CodeGenAndEmitDAG() (./llc+0x2b1ffda)
#12 0x000056500e645c6b llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) (./llc+0x2b22c6b)
#13 0x000056500e648445 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) (.part.0) SelectionDAGISel.cpp:0:0
#14 0x000056500e649c61 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) (./llc+0x2b26c61)
#15 0x000056500d8dc87d llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (.part.0) MachineFunctionPass.cpp:0:0
#16 0x000056500de54481 llvm::FPPassManager::runOnFunction(llvm::Function&) (./llc+0x2331481)
#17 0x000056500de54804 llvm::FPPassManager::runOnModule(llvm::Module&) (./llc+0x2331804)
#18 0x000056500de5632e llvm::legacy::PassManagerImpl::run(llvm::Module&) (./llc+0x233332e)
#19 0x000056500bef5e99 compileModule(char**, llvm::LLVMContext&) llc.cpp:0:0
#20 0x000056500bdf61d6 main (./llc+0x2d31d6)
#21 0x00007ffb1882b2e0 (/lib64/libc.so.6+0x262e0)
#22 0x00007ffb1882b399 __libc_start_main (/lib64/libc.so.6+0x26399)
#23 0x000056500beecab5 _start (./llc+0x3c9ab5)
[1]    20664 IOT instruction  ./llc
```

### Project release version

latest 18 release branch

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Gentoo

### Shell

ZSH

gerekon, 2024-10-21:
Not sure, but it could be related to #91. Or at least it can be fixed with https://github.com/espressif/llvm-project/pull/92

MabezDev, 2024-10-22:
I can confirm it's fixed with #92

closed 2024-10-22 by MabezDev