espressif/llvm-project #104: Support clangd version 19

mrjohannchang, 2024-10-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

C23 will be finalized within a month, most of its features are already available in GCC provided in `esp-13.2.0_20240530` that's bundled with ESP-IDF version 5.3. But the support in the current bundled clangd is a bit behind.

### Use cases

There are some great features in C23, such as `constexpr`. This feature can already be used in ESP-IDF v5.3. However, since it's not supported until clangd version 19, the bundled clangd does not recognize it, leading to errors on every line that contains this syntax. This results in clangd's error messages becoming excessively verbose.

### Alternatives

I can use standard LLVM 19. But without Espressif's patch, some of the includes cannot be identified properly.

gerekon, 2024-10-24:
Hi @mrjohannchang!

We are finishing rebase our fork on LLVM 19 and plan to do the release which also will include clangd.

mrjohannchang, 2024-10-24:
Thank you! I'll close this issue once Espressif's fork is bumped to version 19.

mrjohannchang, 2025-01-09:
Hi @gerekon, want to check if there's any progress of this ticket?

gerekon, 2025-01-10:
Hi @mrjohannchang !

We finished rebasing our release LLVM 19. Had some problems with tests. Now I need to pickup several commits from our LLVM 18 release branch. After that we will make a release. I hope that will be dome soon. Working on it.

mrjohannchang, 2025-02-04:
Happy Lunar New Year @gerekon! I realized though branch v19 is created, the release is not yet made. So I reopened this ticket. I also create a [ticket](https://github.com/espressif/esp-idf/issues/15325) on ESP-IDF for the integration.

gerekon, 2025-02-19:
@mrjohannchang Sorry for the delay. Released https://github.com/espressif/llvm-project/releases/tag/esp-19.1.2_20250211.

Preparing MR for upgrading Clang version in IDF.

mrjohannchang, 2025-02-24:
Thank you @gerekon. The binary does not output version info as expected which causes esp-idf export failing to include this version of clangd.

The output of clang --version this version:

```
Espressif clang version 19.1.2 (https://github.com/espressif/llvm-project.git)
Target: riscv32-esp-unknown-elf
Thread model: posix
InstalledDir: ...
```

Previous version:

```
Espressif clang version 18.1.2 (https://github.com/espressif/llvm-project.git esp-18.1.2_20240912)
Target: riscv32-esp-unknown-elf
Thread model: posix
InstalledDir: ...
```

gerekon, 2025-02-26:
@mrjohannchang Fixed in new release https://github.com/espressif/llvm-project/releases/tag/esp-19.1.2_20250225

mrjohannchang, 2025-03-18:
Thank you @gerekon!

closed 2025-03-18 by mrjohannchang