espressif/llvm-project #110: Highlighting problem on xtensa target with clangd

damian-kurek-wizzdev, 2025-03-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Clangd can highlight both xtensa and RISC-V target

### Actual behavior (suspected bug)

I use QTCreator with clangd provided with this repo, and everything works fine when I compile the target for esp32c6.
When I use the target for xtensa, it does not highlight C++ stl code for example, shared_ptr.
esp32 target
[image]
esp32c6

[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Opening project in QT Creator with clangd running  for esp32 and second one with target for esp32c6

### Project release version

Happens on version 19 and 18

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 20.04

### Shell

other (details in Additional context)

### Additional context

Not sure if it is a bug or something with configuration on my side

gerekon, 2025-03-17:
Hi @damian-kurek-wizzdev 

I am not familiar with QT Creator and neither an appropriate project for it. Could you kindly provide project for testing along with the instructions how to configure QT Creator to use clangd from our distribution.

Another (and much faster) option could be if you'd provided log for clangd for that error.

damian-kurek-wizzdev, 2025-03-18:
Ok,  I figured it out. QTCreator configured compiler for riscV as riscV 32 bits and xtensa as linux 64 bit for some reason. After setting the correct compiler it works perfectly.

closed 2025-03-18 by damian-kurek-wizzdev