espressif/llvm-project #111: Nonmax library does not build

Unaimend, 2025-03-15:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Library gets installed 

### Actual behavior (suspected bug)

Error during compilation
```

  Compiling nonmax v0.5.5
error: failed to parse bitcode for LTO module: Bitwidth for integer type out of range (Producer: 'LLVM18.1.2-rust-1.84.0-nightly' Reader: 'LLVM 18.1.2-rust-1.84.0-nightly')
```

### Error logs or terminal output

```shell
Compiling nonmax v0.5.5
error: failed to parse bitcode for LTO module: Bitwidth for integer type out of range (Producer: 'LLVM18.1.2-rust-1.84.0-nightly' Reader: 'LLVM 18.1.2-rust-1.84.0-nightly')
```

### Steps to reproduce the behavior

Try to build any project with nonmax as dependency (See cargo.toml)
[package]
name = "embedded-proj-std"
version = "0.1.0"
authors = ["Benno Doerr <bencayd@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.77"

[[bin]]
name = "embedded-proj-std"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = []

experimental = ["esp-idf-svc/experimental"]

[dependencies]
log = "0.4"
esp-idf-svc = { version = "0.51", features = ["critical-section", "embassy-time-driver", "embassy-sync"] }
esp-idf-hal = { version = "0.45.2" }
#bevy_ecs = { version = "0.12.1" }
nonmax = "0.5.5"
ssd1306 = "0.9"
embedded-graphics = "0.8.1"
anyhow = "1.0.97"

[build-dependencies]
embuild = "0.33"

### Project release version

LLVM18.1.2-rust-1.84.0-nightl

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04 LTS

### Shell

Bash

gerekon, 2025-03-17:
Hi @Unaimend 

Have you tried to ask our Rust fork guys for help? https://github.com/esp-rs/rust

cc @MabezDev

Unaimend, 2025-03-18:
No. I thought this is the repo to ask llvm questions. But thanks for the pointer. In the meantime, we found out that the problem is related to the optimization level. So if other people have the same issue, Using `s` instead of `z` works.

[image]

gerekon, 2025-03-19:
Ok. I see. Thanks for the findings.
I will try to reproduce the bug when I get my Rust env back. But it can take time since Rust is not my usual dev env.