espressif/llvm-project #112: symbol '.LCPI0_0' is already defined

yamt, 2025-05-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

no errors

### Actual behavior (suspected bug)

while i'm upgrading llvm version for wamr, (https://github.com/bytecodealliance/wasm-micro-runtime/pull/4213)
i observed errors like:
```
spacetanuki% ./wamrc --target=xtensa   --mllvm=-mtext-section-literals --disable
-llvm-lto --bounds-checks=1 -o a ~/git/toywasm/.wasm-spec-test/test/core/address
.3.wasm
Create AoT compiler with:
  target:        xtensa
  target cpu:
  target triple: xtensa-pc-linux-gnu
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file
error: <unknown>:0: symbol '.LCPI0_0' is already defined
```

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

[a.ll.zip](https://github.com/user-attachments/files/19996193/a.ll.zip)
unzip the attached file.
llc -mtext-section-literals -o a.s a.ll
llvm-mc --filetype=obj -o a.o a.s

### Project release version

xtensa_release_19.1.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

15.4.1

### Shell

ZSH

yamt, 2025-07-04:
> [@yamt](https://github.com/yamt) , thank you for investigation and PR , problem is fixed.

thank you for taking care of it.
is there any plan to make a release including the fix?

gerekon, 2025-07-10:
@yamt Are you talking about binary release like https://github.com/espressif/llvm-project/releases/tag/esp-19.1.2_20250312?

We are going to switch to LLVM 20.1.1 rebased branch, so planned to do the next binary release based on it. Do you need one for LLVM 19.1.2?

yamt, 2025-07-11:
> [@yamt](https://github.com/yamt) Or you just need changes to be published on GH branch?

yes, it's enough for me. i don't need binary releases.

also, while i prefer llvm 19 right now, it isn't a strict requirement. llvm 20 is probably ok too.