espressif/llvm-project #115: Using esp-clang to compile will trig CPU0 WDT and/or other strange behavour(s) (EUB-90)

PickaxeHit, 2025-07-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

will run as v5.4(or 5.x)

### Actual behavior (suspected bug)

```log
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x3b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037f59d
--- 0x4037f59d: esp_restart_noos at /home/pickaxehit/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:164
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1550
load:0x403c8700,len:0xf0c
--- 0x403c8700: _stext at ??:?
load:0x403cb700,len:0x3550
entry 0x403c8a9c
--- 0x403c8a9c: call_start_cpu0 at /home/pickaxehit/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25
I (29) boot: ESP-IDF v6.0-dev-1362-g346870a304 2nd stage bootloader
I (29) boot: compile time Jul 27 2025 01:03:32
I (29) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (34) boot: efuse block revision: v1.3
I (37) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 8MB
I (49) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0f6fch ( 63228) map
I (104) esp_image: segment 1: paddr=0001f724 vaddr=3fc99700 size=008f4h (  2292) load
I (105) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=256c4h (153284) map
I (136) esp_image: segment 3: paddr=000456ec vaddr=3fc99ff4 size=02904h ( 10500) load
I (139) esp_image: segment 4: paddr=00047ff8 vaddr=40378000 size=116f0h ( 71408) load
I (157) esp_image: segment 5: paddr=000596f0 vaddr=50000000 size=00020h (    32) load
I (165) boot: Loaded app from partition at offset 0x10000
I (165) boot: Disabling RNG early entropy source...
I (175) cpu_start: Multicore app
I (184) cpu_start: Pro cpu start user code
I (185) cpu_start: cpu freq: 240000000 Hz
I (185) app_init: Application information:
I (185) app_init: Project name:     bridge
I (188) app_init: App version:      v1.1.0-9-g0f04472
I (193) app_init: Compile time:     Jul 27 2025 01:03:30
I (198) app_init: ELF file SHA256:  52aed0aad...
I (203) app_init: ESP-IDF:          v6.0-dev-1362-g346870a304
I (208) efuse_init: Min chip rev:     v0.0
I (212) efuse_init: Max chip rev:     v0.99 
I (216) efuse_init: Chip rev:         v0.2
I (220) heap_init: Initializing. RAM available for dynamic allocation:
I (226) heap_init: At 3FCA1E48 len 000478C8 (286 KiB): RAM
I (231) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (236) heap_init: At 600FE000 len 00001FE8 (7 KiB): RTCRAM
I (242) spi_flash: detected chip: gd
I (245) spi_flash: flash io: dio
I (248) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (254) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (261) main_task: Started on CPU0
I (301) main_task: Calling app_main()
I (301) bridge_main: LED GPIO init done
I (301) bridge_msc: MSC disk RAM usage: 698 bytes
I (301) uart: queue free spaces: 20
I (301) bridge_serial: UART & GPIO have been initialized
I (311) main_task: Returned from app_main()
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x42010ca1  PS      : 0x00060034  A0      : 0x820112b8  A1      : 0x3fc9ab40  
--- 0x42010ca1: dcd_int_handler at /home/pickaxehit/projects/software/esp32/esp-usb-bridge/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1011
A2      : 0x00000000  A3      : 0x00000004  A4      : 0x00000010  A5      : 0x60080014  
A6      : 0xffffffff  A7      : 0x42011024  A8      : 0x800c3814  A9      : 0x04808438  
--- 0x42011024: handle_ep_irq at /home/pickaxehit/projects/software/esp32/esp-usb-bridge/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:953
A10     : 0x00060023  A11     : 0x00000003  A12     : 0x00060023  A13     : 0x3fc9cbb8  
A14     : 0x3fc99820  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x42003ba3  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x42010ca1
--- 0x42003ba3: uart_hal_write_txfifo at /home/pickaxehit/esp/esp-idf/components/hal/uart_hal_iram.c:27
--- 0x42010ca1: dcd_int_handler at /home/pickaxehit/projects/software/esp32/esp-usb-bridge/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1011

Backtrace: 0x42010c9e:0x3fc9ab40 0x420112b5:0x3fc9ab80 0x4037b599:0x3fc9aba0 0x4037c67f:0x3fca5f90 0x4202426c:0x3fca5fb0 0x40383000:0x3fca5fd0 0x4038076d:0x3fca6010
--- 0x42010c9e: dcd_int_handler at /home/pickaxehit/projects/software/esp32/esp-usb-bridge/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1011
--- 0x420112b5: dwc2_int_handler_wrap at /home/pickaxehit/projects/software/esp32/esp-usb-bridge/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dwc2_esp32.h:73
--- 0x4037b599: _xt_lowint1 at /home/pickaxehit/esp/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x4037c67f: xt_utils_wait_for_intr at /home/pickaxehit/esp/esp-idf/components/xtensa/include/xt_utils.h:82
--- (inlined by) esp_cpu_wait_for_intr at /home/pickaxehit/esp/esp-idf/components/esp_hw_support/cpu.c:55
--- 0x4202426c: esp_vApplicationIdleHook at /home/pickaxehit/esp/esp-idf/components/esp_system/freertos_hooks.c:58
--- 0x40383000: prvIdleTask at /home/pickaxehit/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4350
--- 0x4038076d: vPortTaskWrapper at /home/pickaxehit/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

Core  1 register dump:
PC      : 0x4037c682  PS      : 0x00060734  A0      : 0x8202426f  A1      : 0x3fca66f0  
--- 0x4037c682: esp_cpu_wait_for_intr at /home/pickaxehit/esp/esp-idf/components/esp_hw_support/cpu.c:64
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x3fca35d0  A5      : 0x3fca35b0  
A6      : 0x40379f20  A7      : 0x00000001  A8      : 0x820159b5  A9      : 0x3fca66c0  
--- 0x40379f20: ipc_task at /home/pickaxehit/esp/esp-idf/components/esp_system/esp_ipc.c:?
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3fca35b0  A13     : 0x3fca3560  
A14     : 0x3fca1cb0  A15     : 0x3fca68e0  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4037c67f:0x3fca66f0 0x4202426c:0x3fca6710 0x40383000:0x3fca6730 0x4038076d:0x3fca6770
--- 0x4037c67f: xt_utils_wait_for_intr at /home/pickaxehit/esp/esp-idf/components/xtensa/include/xt_utils.h:82
--- (inlined by) esp_cpu_wait_for_intr at /home/pickaxehit/esp/esp-idf/components/esp_hw_support/cpu.c:55
--- 0x4202426c: esp_vApplicationIdleHook at /home/pickaxehit/esp/esp-idf/components/esp_system/freertos_hooks.c:58
--- 0x40383000: prvIdleTask at /home/pickaxehit/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4350
--- 0x4038076d: vPortTaskWrapper at /home/pickaxehit/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 52aed0aad

Rebooting...

```

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. use IDF master 
```
$ git rev-parse HEAD
346870a3044010f2018be0ef3b86ba650251c655
```
2. build and flash
3. connect UART0 to PC

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Arch 

### Shell

Bash

andreisfr, 2025-12-25:
Hello @PickaxeHit , @Dzarda7  ! Thank you for reporting the problem. May I ask you a question about which board configuration was used to reproduce the problem? You used esp32s3 board for bridge with connected esp32 board? Does problem appear without esp32 board connected?

Dzarda7, 2025-12-25:
Hello @andreisfr, thanks for looking into this. I just tested and steps I did to reproduce are:

1. Checkout release/v5.5 branch of ESP-IDF, exact commit is `366dadf3543c592ce806f33cecc4606281d2a684`, but I believe the issue is everywhere as I was able to reproduce on v5.3 previously too.
2. Run `idf_tools.py install esp-clang`
3. `export IDF_TOOLCHAIN=clang`
4. Clone [esp-usb-bridge](https://github.com/espressif/esp-usb-bridge)
5. Run `idf.py set-target esp32s3`
6. Connect [ESP32-S3-DevKitC-1 v1.1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide_v1.1.html) through UART port to the PC
7. Run `idf.py build flash monitor`, at this point still everything works, nothing else is connected to the board
8. Connect also USB port to the PC and here Watchdog starts to get triggered.

Tested with `esp-19.1.2_20250312` on Fedora. I believe ESP-IDF version, nor OS, nor devkit makes a difference. Feel free to ask additional questions.

andreisfr, 2025-12-30:
@Dzarda7 , thank you very much. I'm able to reproduce the problem and investigating it.

Dzarda7, 2026-03-11:
@andreisfr thanks a lot for the investigation, will try to address that. In that it probably does not work for the whole esp-usb. cc: @tore-espressif